britkat1980/ha-addons #1: New user. Can I help with feedback?

LeeFarm11, 2024-05-15:
Hi. I have installed Beta and I have some data. 3 phase Gen 3 inverter plus 6 batteries. Some of the numbers look wrong - Battery Power readings seem 10x too high. Many errors in Log. How can I support you with data?
[image]
[image]
[image]

britkat1980, 2024-05-15:
Thanks, let me know what data points seem wrong and I'll look into them.

Regarding the errors, it seems that the inverter network comms is not stable. Perhaps the vast amounts of register sets requires a longer (than 3s) timeout period?

LeeFarm11, 2024-05-16:
Hi. Sorry I can't comment on the timeout. This is very new to me. I stopped the GivTCP add-on and run only this beta add-on. Is that correct? 
Is there some way I can send you all the data you need in one hit?
There is loads of Battery data also now appeared in the Overview, but all the reading are 0.0C or 0V for all 24 cells in all 6 batteries.
[image]
Here are some examples where the Battery Energy readings are 10x too high.
Battery Charge Energy Total kWh 17087 in HA, 1709 in Giv Portal
[image]
Battery Charge Energy Today kWh 220 in HA, 22.4 in Giv Portal
[image]
Battery Discharge Energy Today kWh 133 in HA, 13.3 in Giv Portal
[image]
[image]

britkat1980, 2024-05-16:
OK, I've added in the (missing) 0.1 scaling factor for the energy stats, try 2.4.261 and see if that's better

LeeFarm11, 2024-05-16:
Hi. Thanks.
I uninstalled BETA. I installed DEV 2.4.262
Now the kWh readings look better. 
I do not have GivTCP running. I only have GivTCP-DEV running.

But -
Now I have double of everything. I have Entities starting with 'GivTCP' - correct. AND I have full set of entities starting with 'GivInverter' and they are all 'Unavailable'. For example here is the Device 'GivTCP GivTCP Stats'. Note I do not have version 2.4.255 installed.
[image]

britkat1980, 2024-05-17:
You need to make sure that your “HA Prefix” config item is identical in both dev and beta. It defaults to GivTCP. My guess is you changed it to GivInverter on the Beta, but not the Dev.
Change it and then it will match. You will need to delete the old entities once you have done that

LeeFarm11, 2024-05-17:
I only have GivTCP-DEV running. Not GivTCP and not GivTCP-BETA. Yes when I was running BETA I did change the default name, but after uninstalling, I would have thought the entities would disappear. Lesson learnt. 
427 entities now removed.

I do still have 288 Battery entities showing 0.0C or 0.0V
[image]

britkat1980, 2024-05-17:
Ok, let me see if I can look into the zero battery data. How many batteries do you have?

LeeFarm11, 2024-05-17:
Hi. 6 batteries.

Date: Friday, 17 May 2024 at 08:29
Cc: LeeFarm11 ***@***.***>, Author ***@***.***>

closed 2024-08-27 by britkat1980