britkat1980/ha-addons #3: GivTCP Inverter Details Dev 2.4.270

LeeFarm11, 2024-05-31:
Hi.
In the GivTCP Inverter Details section I have some strange readings. I have a few Unknowns -

GivTCP Battery Priority
GivTCP DC Status
GivTCP Inverter Time
GivTCP System Mode

GivTCP Inverter Firmware is showing the wrong version. Showing D0.0-A0.0 but I have
[image]

GivTCP Inverter Type shows as Gen 1 - Hybrid_3ph but I believe it is Gen 3.

britkat1980, 2024-06-01:
Given the number of issues it would be great to connect directly to troubleshoot, are you able to setup a VPN and allow me access?? If you're in the fb group then DM me there

LeeFarm11, 2024-06-11:
In 2.4.284 this section - GivTCP Inverter Details - has disappeared.

britkat1980, 2024-06-11:
It’s now under the serial number of the inverter, matching the new naming convention for batteries

LeeFarm11, 2024-06-11:
Ah. Thanks. Yes I found it. 

Still have the same 4 Unknowns.
Software version still shows D0.0-A0.0
Inverter Type still shows Gen1

closed 2024-08-27 by britkat1980