britkat1980/ha-addons #10: How do I Install GivTCP Beta in HA?

Croft5, 2024-08-17:
I can install the normal GivTCP in HA fine. How do I install the Beta version?
I have joined the Beta channel in HA.

Thanks

britkat1980, 2024-08-17:
If you have added this repo into the Addon store in HA.
Then just select the Beta addon, install it and then stop the “old” version. Start this one then head to the web page to configure it

Croft5, 2024-08-17:
Hi,

I am not seeing a beta version to select...

***@***.***

I can see beta versions for other programs but not for GivTCP???

Dave

britkat1980, 2024-08-17:
So you don’t see this in the addon store?
[image]

Have you added the repo here:
[image]

Croft5, 2024-08-17:
I am using a W11 desktop rather than a phone, but no I do not see what you sent.

I go to the 3 dots in the Add-on Store and can find this repository...

***@***.***

I add that and all I get is the standard GivTCP option, no beta or dev

Dave

IMG_5419.jpeg (view on web)<https://github.com/user-attachments/assets/675abdc7-22ad-4cf7-ae63-92e3882d24c7>

IMG_5421.jpeg (view on web)<https://github.com/user-attachments/assets/32404c6a-2044-4d6b-9c28-d8106b0bad99>

-

LeeFarm11, 2024-08-17:
You need to add this repository -

https://github.com/britkat1980/ha-addons

Croft5, 2024-08-17:
Yes, that worked, thanks so much

https://github.com/britkat1980/ha-addons

-

LeeFarm11, 2024-08-17:
Welcome aboard. I'm hoping to get up and running with Predbat, but need to get GivTCP working first with the 3 phase Inverter. Seems many teething issues with the 3 phase but hopefully eventually everything will fall into place. Do you have a 3 phase Inverter?

Croft5, 2024-08-17:
No, one phase is enough 😊
All the very best with your setup.
I have just installed GivTCP, now for the setup

Croft5, 2024-08-17:
All working now 😊

Thanks, Dave

closed 2024-08-27 by britkat1980