britkat1980/ha-addons #13: Install GivTCP Beta problem

plug-it-in, 2024-08-24:
Because of issues Ive been having I decided to do a totally new install on a new HA image.

So from new HA image I just added the haaddon repository and only installed GivTCP Beta.

The log shows this
2024-08-24 20:05:12,593 - startup - [INFO] - ========================== STARTING GivTCP================================
2024-08-24 20:05:12,602 - startup - [INFO] - No HA MQTT service has been found. Install and run the Mosquitto addon, or manually configure your own MQTT broker.
2024-08-24 20:05:14,631 - startup - [INFO] - Searching for Inverters
2024-08-24 20:05:14,632 - startup - [INFO] - Scanning network for GivEnergy Devices...
2024-08-24 20:05:29,057 - startup - [INFO] - Inverter GW2330G363 which is a Gen1 - Gateway with 0 batteries has been found at: 192.168.1.73
2024-08-24 20:05:39,255 - startup - [INFO] - Inverter CH2335G038 which is a Gen1 - All_in_one with 4 batteries has been found at: 192.168.1.72
Traceback (most recent call last):
  File "/app/startup.py", line 433, in <module>
    shutil.copyfile("settings.json",SFILE)
  File "/usr/local/lib/python3.11/shutil.py", line 258, in copyfile
    with open(dst, 'wb') as fdst:
         ^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '/config/GivTCP/allsettings.json'

I have seen this consistently that the GivTCP folder is not created during install. On my older image I installed GivTCP and this does create this folder. But I dont want to do this on this image as it seems like a problem. 

I will instal MQTT and see if this does anything and feedback.

plug-it-in, 2024-08-24:
OK no further forward after MQTT install same error about allsettings file and directory missing. How should I fix this? (see log)

2024-08-24 20:17:58,328 - startup - [INFO] - ========================== STARTING GivTCP================================
2024-08-24 20:18:00,364 - startup - [INFO] - Searching for Inverters
2024-08-24 20:18:00,364 - startup - [INFO] - Scanning network for GivEnergy Devices...
2024-08-24 20:18:24,130 - startup - [INFO] - Inverter CH2335G008 which is a Gen1 - All_in_one with 4 batteries has been found at: 192.168.1.71
2024-08-24 20:18:30,414 - startup - [INFO] - Inverter CH2335G038 which is a Gen1 - All_in_one with 4 batteries has been found at: 192.168.1.72
2024-08-24 20:18:31,592 - startup - [INFO] - Inverter GW2330G363 which is a Gen1 - Gateway with 0 batteries has been found at: 192.168.1.73
Traceback (most recent call last):
  File "/app/startup.py", line 433, in <module>
    shutil.copyfile("settings.json",SFILE)
  File "/usr/local/lib/python3.11/shutil.py", line 258, in copyfile
    with open(dst, 'wb') as fdst:
         ^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '/config/GivTCP/allsettings.json'

plug-it-in, 2024-08-24:
Just as final evidence that GivTCP is not created
[image]

BondBug, 2024-08-27:
If you manually create a empty folder GivTCP then you will find that gGivTCP Beta will install fine.I had the same issue on my Raspberry Pi and found that by creating a empty Folder  GivTCP then the beta will install no problem

britkat1980, 2024-08-27:
If you are installing via docker, I suspect it needs /config in place, which for the HA addon is given as a mapped volume. This allows the config files to be persistent and available.

Is this the same experience you are seeing?

plug-it-in, 2024-08-27:
Not sure how to answer this. I'm just using raspberry pi with HA image.
Then in HA I add your repository and then install givtcp beta and you see
the result. Seems from the other comment you've missed the check for the
Givtcp directory and creating it if it does exist during install.

britkat1980, 2024-08-27:
This is the first time I’ve had a report of this error and have lots of users running on a rpi as an addon. Can you confirm exactly what directory you need to create and its absolute path, so I can check?
Thanks

BondBug, 2024-08-27:
In Homeassistant use  File Editor addon  and in the Homeassistant Directory create a Folder Called GivTCP with capitals as shown then retry installing Givtcp beta and it will install no problem

BondBug, 2024-08-27:
you will find that if you have already installed the standard GivTCP then the beta version will install no problem as when you unistall that version it leaves the GivTCP folder intact

britkat1980, 2024-08-27:
Ok, thanks I’ll make sure I put a check in place for presence of the folder and create if needed

BondBug, 2024-08-27:
Just to let you know I have been running GivTCP Beta 698 with no problems since realeased but I did have to disable the battery pause charge and discharge settings in the Prebat Apps Yaml once this was done I have no errors in prebat or GivTCP

BondBug, 2024-08-27:
forgot to say I am using a AC3 inverter

britkat1980, 2024-08-27:
Battery charge rate issue with predbat is now fixed in latest dev builds

BondBug, 2024-08-27:
brill great i think I will wait till you put in the Beta cheers thanks for all your hard work

BondBug, 2024-08-27:
[image]

BondBug, 2024-08-27:
GivTCP director Location above

britkat1980, 2024-08-28:
Try Dev722 and see if it fixes this issue

BondBug, 2024-08-28:
Hi unistalled beta and deleted GivTCP Directory and tried to install Dev  2.4.723 but install failed see below and no GivTCP directory created sorry for the bad news

2024-08-27 00:07:51,519 - startup - [INFO] - ========================== STARTING GivTCP================================
2024-08-27 00:07:53,532 - startup - [INFO] - Searching for Inverters
2024-08-27 00:07:53,532 - startup - [INFO] - Scanning network for GivEnergy Devices...
2024-08-27 00:08:05,399 - startup - [INFO] - Inverter CE2223G363 which is a Gen1 - Ac with 2 batteries has been found at: 192.168.1.111
2024-08-27 00:08:05,408 - startup - [INFO] - Inverter CE2223G363 not in settings file
2024-08-27 00:08:05,409 - startup - [INFO] - Adding Inverter CE2223G363 to slot 1
2024-08-27 00:13:07,800 - startup - [INFO] - ========================== STARTING GivTCP================================
2024-08-27 00:13:09,813 - startup - [INFO] - Searching for Inverters
2024-08-27 00:13:09,813 - startup - [INFO] - Scanning network for GivEnergy Devices...
2024-08-27 00:13:23,497 - startup - [INFO] - Inverter CE2223G363 which is a Gen1 - Ac with 2 batteries has been found at: 192.168.1.111
2024-08-27 00:13:23,507 - startup - [INFO] - Inverter CE2223G363 not in settings file
2024-08-27 00:13:23,507 - startup - [INFO] - Adding Inverter CE2223G363 to slot 1
Traceback (most recent call last):
  File "/app/startup.py", line 29, in <module>
    fh = TimedRotatingFileHandler("/config/GivTCP/startup.log", when='midnight', backupCount=7)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/logging/handlers.py", line 214, in __init__
    BaseRotatingHandler.__init__(self, filename, 'a', encoding=encoding,
  File "/usr/local/lib/python3.11/logging/handlers.py", line 58, in __init__
    logging.FileHandler.__init__(self, filename, mode=mode,
  File "/usr/local/lib/python3.11/logging/__init__.py", line 1181, in __init__
    StreamHandler.__init__(self, self._open())
                                 ^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/logging/__init__.py", line 1213, in _open
    return open_func(self.baseFilename, self.mode,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '/config/GivTCP/startup.log'

BondBug, 2024-08-28:
Tried to install DEV  2.4.724 with manually created GivTCP Directory but then goth the below
 File "/app/startup.py", line 403, in <module>
    finv=findinv(networks)
  File "/app/startup.py", line 292, in findinv
    deets=asyncio.run(getInvDeets(invList[inv]))
  File "/usr/local/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
  File "/usr/local/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 637, in run_until_complete
    self.run_forever()
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 604, in run_forever
    self._run_once()
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 1909, in _run_once
    handle._run()
  File "/usr/local/lib/python3.11/asyncio/events.py", line 80, in _run
    self._context.run(self._callback, *self._args)
  File "/app/startup.py", line 102, in getInvDeets
    await client.detect_plant(additional=False)
  File "/app/GivTCP/givenergy_modbus_async/client/client.py", line 239, in detect_plant
    _logger.info("Batteries detected: %d", self.plant.bcu_list)
Message: 'Batteries detected: %d'
Arguments: ([],)
--- Logging error ---
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/logging/__init__.py", line 1110, in emit
    msg = self.format(record)
          ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/logging/__init__.py", line 953, in format
    return fmt.format(record)
           ^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/logging/__init__.py", line 687, in format
    record.message = record.getMessage()
                     ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/logging/__init__.py", line 377, in getMessage
    msg = msg % self.args
          ~~~~^~~~~~~~~~~
TypeError: %d format: a real number is required, not list
Call stack:
  File "/app/startup.py", line 403, in <module>
    finv=findinv(networks)
  File "/app/startup.py", line 292, in findinv
    deets=asyncio.run(getInvDeets(invList[inv]))
  File "/usr/local/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
  File "/usr/local/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 637, in run_until_complete
    self.run_forever()
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 604, in run_forever
    self._run_once()
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 1909, in _run_once
    handle._run()
  File "/usr/local/lib/python3.11/asyncio/events.py", line 80, in _run
    self._context.run(self._callback, *self._args)
  File "/app/startup.py", line 102, in getInvDeets
    await client.detect_plant(additional=False)
  File "/app/GivTCP/givenergy_modbus_async/client/client.py", line 240, in detect_plant
    _logger.info("Meters detected: %d", self.plant.meter_list)
Message: 'Meters detected: %d'
Arguments: ([],)
--- Logging error ---
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/logging/__init__.py", line 1110, in emit
    msg = self.format(record)
          ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/logging/__init__.py", line 953, in format
    return fmt.format(record)
           ^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/logging/__init__.py", line 687, in format
    record.message = record.getMessage()
                     ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/logging/__init__.py", line 377, in getMessage
    msg = msg % self.args
          ~~~~^~~~~~~~~~~
TypeError: %d format: a real number is required, not list
Call stack:
  File "/app/startup.py", line 403, in <module>
    finv=findinv(networks)
  File "/app/startup.py", line 292, in findinv
    deets=asyncio.run(getInvDeets(invList[inv]))
  File "/usr/local/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
  File "/usr/local/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 637, in run_until_complete
    self.run_forever()
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 604, in run_forever
    self._run_once()
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 1909, in _run_once
    handle._run()
  File "/usr/local/lib/python3.11/asyncio/events.py", line 80, in _run
    self._context.run(self._callback, *self._args)
  File "/app/startup.py", line 102, in getInvDeets
    await client.detect_plant(additional=False)
  File "/app/GivTCP/givenergy_modbus_async/client/client.py", line 240, in detect_plant
    _logger.info("Meters detected: %d", self.plant.meter_list)
Message: 'Meters detected: %d'
Arguments: ([],)
2024-08-28 16:44:46,262 - Startup - client      -  [INFO    ] - Slave address in use: 49
2024-08-28 16:44:46,263 - Startup - startup     -  [INFO    ] - Inverter CE2223G363 which is a Gen1 - Ac with 2 batteries has been found at: 192.168.1.111
2024-08-28 16:44:46,273 - Startup - startup     -  [INFO    ] - Inverter CE2223G363 already found in settings file (slot 1), checking IP address is unchanged...

BondBug, 2024-08-28:
Ok reinstalled Beta having manually creating GivTCP directory all good no problems then stopped beta and install DEV Version and this installed and started correctly having not touched the existing GivTCP Directory created bt the Beta version very strange ! see bleow
[2024-08-28 17:11:01 +0100] [80] [INFO] Starting gunicorn 22.0.0
[2024-08-28 17:11:01 +0100] [80] [INFO] Listening at: http://0.0.0.0:6345 (80)
[2024-08-28 17:11:01 +0100] [80] [INFO] Using worker: sync
[2024-08-28 17:11:01 +0100] [81] [INFO] Booting worker with pid: 81
[2024-08-28 17:11:01 +0100] [82] [INFO] Booting worker with pid: 82
[2024-08-28 17:11:01 +0100] [84] [INFO] Booting worker with pid: 84
2024-08-28 17:11:01,955 - Inv1 - read        -  [INFO    ] - Starting watch_plant loop...
2024-08-28 17:11:01,957 - Inv1 - read        -  [CRITICAL] - Detecting inverter characteristics...
2024-08-28 17:11:11,242 - Inv1 - read        -  [INFO    ] - Publishing Home Assistant Discovery messages

britkat1980, 2024-08-28:
Once the GivTCP directory exists it’s used by both Beta and Dev, allowing you to switch between the two.
Can I confirm that the dir was created automatically by GivTCP?

BondBug, 2024-08-28:
Sorry should of explained myself better in my prior text which was misleading I had to manually create the GIvTCP directory before the Beta version would run

BondBug, 2024-08-28:
same error message as the user Plug-it-in got when trying to istall without a GivTCP directory existing

BondBug, 2024-08-28:
For Info when you unistall any version of GivTCP it does not remove the GivTCP directory or any of its contents in raspberrypi Home Assistant Operating System   so to check clean install you have to manually delete the GivTCP directory first to do the install test on Beta or Dev Vesions

plug-it-in, 2024-08-29:
No the Givtcp directory was not created by givtcp I had to do it manually

plug-it-in, 2024-08-29:
All worked for me if I created directory manually before the install.

BondBug, 2024-08-29:
Hi Mark just tried again to install Dev with the GivTCP directory already manually created but no Beta installed  and got the below  log and then no more and if I tried to run the config I get this message (  Error: SyntaxError: JSON.parse: unexpected character at line 1 column 1 of the JSON data ) If I then install the Beta version and run the config followed by installing the Dev version the Dev works no problem. In either case the GivTCP directory has to be manually created first.

2024-08-29 11:16:18,564 - Startup - startup     -  [INFO    ] - ========================== STARTING GivTCP================================
2024-08-29 11:16:20,578 - Startup - startup     -  [INFO    ] - Searching for Inverters
2024-08-29 11:16:20,578 - Startup - startup     -  [INFO    ] - Scanning network for GivEnergy Devices...
2024-08-29 11:16:32,950 - Startup - client      -  [INFO    ] - Connection established to 192.168.1.111:8899
2024-08-29 11:16:32,951 - Startup - client      -  [INFO    ] - Detecting plant
2024-08-29 11:16:34,187 - Startup - client      -  [INFO    ] - Plant Detected
2024-08-29 11:16:36,947 - Startup - client      -  [INFO    ] - Batteries detected: 2
2024-08-29 11:16:36,947 - Startup - client      -  [INFO    ] - Meters detected: 0
2024-08-29 11:16:36,947 - Startup - client      -  [INFO    ] - Slave address in use: 49
2024-08-29 11:16:36,948 - Startup - startup     -  [INFO    ] - Inverter CE2223G363 which is a Gen1 - Ac with 2 batteries has been found at: 192.168.1.111
2024-08-29 11:16:36,957 - Startup - startup     -  [INFO    ] - Inverter CE2223G363 not in settings file
2024-08-29 11:16:36,958 - Startup - startup     -  [INFO    ] - Adding Inverter CE2223G363 to slot 1

BondBug, 2024-08-29:
Hi Mark tried the same process on a Mac Min running Homeassistant in a VM  and getting the same problems as I had on my Raspberry pi please see below The only way I cound get the Dev version to work was by installing and running the Beta version first 

**Error from Dev when no GivTCP Folder**

Traceback (most recent call last):
  File "/app/startup.py", line 291, in <module>
    fh = TimedRotatingFileHandler("/config/GivTCP/startup.log", when='midnight', backupCount=7)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/logging/handlers.py", line 214, in __init__
    BaseRotatingHandler.__init__(self, filename, 'a', encoding=encoding,
  File "/usr/local/lib/python3.11/logging/handlers.py", line 58, in __init__
    logging.FileHandler.__init__(self, filename, mode=mode,
  File "/usr/local/lib/python3.11/logging/__init__.py", line 1181, in __init__
    StreamHandler.__init__(self, self._open())
                                 ^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/logging/__init__.py", line 1213, in _open
    return open_func(self.baseFilename, self.mode,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '/config/GivTCP/startup.log'

When Dev Installed after manually creating a GivTCP Folder 

GivTCP================================
2024-08-29 12:49:54,205 - Startup - startup     -  [INFO    ] - Searching for Inverters
2024-08-29 12:49:54,205 - Startup - startup     -  [INFO    ] - Scanning network for GivEnergy Devices...
2024-08-29 12:50:07,498 - Startup - client      -  [INFO    ] - Connection established to 192.168.1.111:8899
2024-08-29 12:50:07,498 - Startup - client      -  [INFO    ] - Detecting plant
2024-08-29 12:50:08,735 - Startup - client      -  [INFO    ] - Plant Detected
2024-08-29 12:50:11,508 - Startup - client      -  [INFO    ] - Batteries detected: 2
2024-08-29 12:50:11,508 - Startup - client      -  [INFO    ] - Meters detected: 0
2024-08-29 12:50:11,508 - Startup - client      -  [INFO    ] - Slave address in use: 49
2024-08-29 12:50:11,509 - Startup - startup     -  [INFO    ] - Inverter CE2223G363 which is a Gen1 - Ac with 2 batteries has been found at: 192.168.1.111
2024-08-29 12:50:11,512 - Startup - startup     -  [INFO    ] - Inverter CE2223G363 not in settings file
2024-08-29 12:50:11,513 - Startup - startup     -  [INFO    ] - Adding Inverter CE2223G363 to slot 1

But Config will not Run same error message  Error: SyntaxError: JSON.parse: unexpected character at line 1 column 1 of the JSON data
Although  allsettings.json and log_inv_1.log are created in the Manually created GivTCP Directory 

Error from Beta with No GivTCP Folder below but installed correctly after creating the GivTCP Folder b

24-08-29 12:47:19,796 - startup - [INFO] - ========================== STARTING GivTCP================================
2024-08-29 12:47:21,804 - startup - [INFO] - Searching for Inverters
2024-08-29 12:47:21,804 - startup - [INFO] - Scanning network for GivEnergy Devices...
2024-08-29 12:47:37,342 - startup - [INFO] - Inverter CE2223G363 which is a Gen1 - Ac with 2 batteries has been found at: 192.168.1.111
Traceback (most recent call last):
  File "/app/startup.py", line 433, in <module>
    shutil.copyfile("settings.json",SFILE)
  File "/usr/local/lib/python3.11/shutil.py", line 258, in copyfile
    with open(dst, 'wb') as fdst:
         ^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '/config/GivTCP/alls

britkat1980, 2024-08-29:
What dev version are you using? I added a line to create the GivTCP folder as the first step in the startup file. Are you running the latest dev version (beta doesn’t have this in yet)

BondBug, 2024-08-29:
yes latest version : 2.4.728

britkat1980, 2024-08-29:
Confirmed and I found the issue. Dev734 should now correctly create the GivTCP folder (I was missing a single "/" from the path!)

BondBug, 2024-08-29:
HI Mark Latest Dev Version created GivTCP Folder (with startup.log and allsettings.json included) and the Allsettings had my inverter correctly identified with correct Ip address etc.
Only problem now is the Config Page button is still reporting error 
Error: SyntaxError: JSON.parse: unexpected character at line 1 column 1 of the JSON data allsettings.json

BondBug, 2024-08-29:
Start Log file below
2024-08-29 14:53:16,639 - Startup - startup     -  [INFO    ] - ========================== STARTING GivTCP================================
2024-08-29 14:53:18,652 - Startup - startup     -  [INFO    ] - Searching for Inverters
2024-08-29 14:53:18,652 - Startup - startup     -  [INFO    ] - Scanning network for GivEnergy Devices...
2024-08-29 14:53:31,935 - Startup - client      -  [INFO    ] - Connection established to 192.168.1.111:8899
2024-08-29 14:53:31,936 - Startup - client      -  [INFO    ] - Detecting plant
2024-08-29 14:53:33,173 - Startup - client      -  [INFO    ] - Plant Detected
2024-08-29 14:53:35,933 - Startup - client      -  [INFO    ] - Batteries detected: 2
2024-08-29 14:53:35,934 - Startup - client      -  [INFO    ] - Meters detected: 0
2024-08-29 14:53:35,934 - Startup - client      -  [INFO    ] - Slave address in use: 49
2024-08-29 14:53:35,934 - Startup - startup     -  [INFO    ] - Inverter CE2223G363 which is a Gen1 - Ac with 2 batteries has been found at: 192.168.1.111
2024-08-29 14:53:35,944 - Startup - startup     -  [INFO    ] - Inverter CE2223G363 not in settings file
2024-08-29 14:53:35,944 - Startup - startup     -  [INFO    ] - Adding Inverter CE2223G363 to slot 1

britkat1980, 2024-08-29:
Can you open the json file and screenshot it? Two users reporting this now, but not sure why!

>
> HI Mark Latest Dev Version created GivTCP Folder (with startup.log and allsettings.json included) and the Allsettings had my inverter correctly identified with correct Ip address etc.
Only problem now is the Config Page button is still reporting error
Error: SyntaxError: JSON.parse: unexpected character at line 1 column 1 of the JSON data allsettings.json

BondBug, 2024-08-29:
See  below line 1 has open { 

{
    "v3upgrade": true,
    "invertorIP_1": "192.168.1.111",
    "inverter_enable_1": false,
    "Model_1": "Ac",
    "serial_number_1": "CE2223G363",
    "inverterName_1": "GivTCP",
    "invertorIP_2": "",
    "inverter_enable_2": false,
    "Model_2": "",
    "serial_number_2": "",
    "inverterName_2": "GivTCP2",
    "invertorIP_3": "",
    "inverter_enable_3": false,
    "Model_3": "",
    "serial_number_3": "",
    "inverterName_3": "GivTCP3",
    "invertorIP_4": "",
    "inverter_enable_4": false,
    "Model_4": "",
    "serial_number_4": "",
    "inverterName_4": "GivTCP4",
    "invertorIP_5": "",
    "inverter_enable_5": false,
    "Model_5": "",
    "serial_number_5": "",
    "inverterName_5": "GivTCP5",
    "MQTT_Output": true,
    "MQTT_Address": "core-mosquitto",
    "MQTT_Username": "addons",
    "MQTT_Password": "Aotoopaiw5Ohbic3Ochieng9Wu4Aj6lahTiengeengaze3cuing5eeng1nohK9gi",
    "MQTT_Retain": true,
    "cache_location": "/config/GivTCP",
    "TZ": "Europe/London",
    "MQTT_Topic": "",
    "MQTT_Port": 1883,
    "Log_Level": "Info",
    "Print_Raw_Registers": true,
    "self_run": false,
    "self_run_timer": 30,
    "self_run_timer_full": 120,
    "queue_retries": 2,
    "data_smoother": "medium",
    "Influx_Output": false,
    "influxURL": "",
    "influxToken": "",
    "influxBucket": "",
    "influxOrg": "",
    "HA_Auto_D": true,
    "PYTHONPATH": "/app",
    "dynamic_tariff": false,
    "day_rate": 0.395,
    "night_rate": 0.155,
    "export_rate": 0.04,
    "day_rate_start": "05:30",
    "night_rate_start": "23:30",
    "Web_Dash": false,
    "Web_Dash_Port": 3000,
    "Smart_Target": false,
    "GE_API": "",
    "SOLCASTAPI": "",
    "SOLCASTSITEID": "",
    "SOLCASTSITEID2": "",
    "PALM_WINTER": "01,02,03,10,11,12",
    "PALM_SHOULDER": "04,05,09",
    "PALM_MIN_SOC_TARGET": 25,
    "PALM_MAX_SOC_TARGET": 45,
    "PALM_BATT_RESERVE": 4,
    "PALM_BATT_UTILISATION": 0.85,
    "PALM_WEIGHT": 35,
    "LOAD_HIST_WEIGHT": "1",
    "evc_ip_address": "",
    "serial_number_evc": "",
    "evc_enable": false,
    "evc_self_run_timer": 10,
    "evc_import_max_current": 60
}

BondBug, 2024-08-29:
Hi Mark manually edited the allsettings.json to set self run to true and I am getting below in log file and stuck
2024-08-29 15:25:03,693 - Startup - startup     -  [INFO    ] - ========================== STARTING GivTCP================================
2024-08-29 15:25:05,706 - Startup - startup     -  [INFO    ] - Searching for Inverters
2024-08-29 15:25:05,706 - Startup - startup     -  [INFO    ] - Scanning network for GivEnergy Devices...
2024-08-29 15:25:16,011 - Startup - client      -  [INFO    ] - Connection established to 192.168.1.111:8899
2024-08-29 15:25:16,011 - Startup - client      -  [INFO    ] - Detecting plant
2024-08-29 15:25:17,254 - Startup - client      -  [INFO    ] - Plant Detected
2024-08-29 15:25:20,006 - Startup - client      -  [INFO    ] - Batteries detected: 2
2024-08-29 15:25:20,006 - Startup - client      -  [INFO    ] - Meters detected: 0
2024-08-29 15:25:20,006 - Startup - client      -  [INFO    ] - Slave address in use: 49
2024-08-29 15:25:20,007 - Startup - startup     -  [INFO    ] - Inverter CE2223G363 which is a Gen1 - Ac with 2 batteries has been found at: 192.168.1.111
2024-08-29 15:25:20,017 - Startup - startup     -  [INFO    ] - Inverter CE2223G363 already found in settings file (slot 1), checking IP address is unchanged...

closed 2024-08-29 by britkat1980