britkat1980/ha-addons #16: Upgrading from 2.4.x to v3. Advice please.

mprowe, 2024-09-10:
I am running Home Assistant with GivTCP Addon installed. Over the last couple of upgrades I have noticed this comment in the Change Log:
`Future updates from v3 will be pushed to https://github.com/britkat1980/ha-addons.
Please add this to your Home Assistant addon Store to get the upcoming v3 release`

Checking in my Add-On store I see this:
[image]
I this good enough? Will an update happen automaticly at some time? Or do I need to "press a button"?

And should I?  Is v3 for everyone? Or is it case dependant?

Regards, Martin.

britkat1980, 2024-09-10:
Yes this is all ready for v3. Whether you decide to upgrade or not depends on what inverter you have. Some is only support via v3. However there are a huge amount of performance benefits and stability improvements in v3, and I would recommend updating.

>
> I am running Home Assistant with GivTCP Addon installed. Over the last couple of upgrades I have noticed this comment in the Change Log:
> Future updates from v3 will be pushed to https://github.com/britkat1980/ha-addons. Please add this to your Home Assistant addon Store to get the upcoming v3 release
>
>
> Checking in my Add-On store I see this:
> image.png (view on web) (https://github.com/user-attachments/assets/45c3a662-6275-43e9-8bc9-aef6562af721)
I this good enough? Will an update happen automaticly at some time? Or do I need to "press a button"?

mprowe, 2024-09-10:
Okay. Happy to "upgrade"... But I can't find a button?
Do I need to Uninstall v2.4.9 first AND then install v3?
Regards, M.

britkat1980, 2024-09-10:
v3 isn't out just yet, update to 2.4.9 then when v3 is ready you'll get the message

britkat1980, 2024-09-21:
v3 out now

closed 2024-09-21 by britkat1980

mprowe, 2024-09-23:
Do I need to uninstall 2.4.9 first?
Will the sensor IDs remain the same or will I need to redo my Automations?

gcoan, 2024-09-23:
> Do I need to uninstall 2.4.9 first?
> 
> Will the sensor IDs remain the same or will I need to redo my Automations?

You don't need to uninstall 2.4.9, I have it still installed alongside v3 and have swapped between the two several times.

The sensors are all the same names so no dashboard or automation changes required although I did find an issue with the battery sensors not working in v3.  I believe there is a fix to this in 3.0.1 but I have not tried it yet

gcoan, 2024-09-23:
> > I have it still installed alongside v3 and have swapped between the two several times.
> 
> "Swapped"? How do you do that? 

I simply shutdown the v2 add-on and start the v3 add-on, they are separate add-on's in the add-on store.  Just need to make sure that you tick/untick the start on boot options appropriately, and as I use Predbat I have to change the auto-restart config that restarts GivTCP if predbat detects a comms failure (although v3 seems to detect such failures itself better)

gcoan, 2024-09-25:
With v3 you don't need to add your existing MQTT user to GivTCP (as you had to do with v2).  Just leave the fields as they are and GivTCP v3 will auto populate.

Nothing needed to be changed in MQTT either for v3 to work on my HA

mprowe, 2024-09-25:
Thanks, and yes, given time (30mins?) the numbers are showing up in the MQTT Explorer.
Next issue is the Tariff settings in the Config page.
1. Are the Day Start/Night Start UTC or local?
2. Where is the Export rate defined?
Regards, M.

britkat1980, 2024-09-25:
Day/Night times are timezone aware, so they will be local time.
Export limit is defined by installer at commission time and is read only

mprowe, 2024-09-25:
Alert!!!!
Just upgraded to 3.0.2 .... and it won't run!
` HTTP  9/25/2024 1:00:46 PM 127.0.0.1 GET /js/converters.js
 HTTP  9/25/2024 1:00:46 PM 127.0.0.1 Returned 304 in 8 ms
 HTTP  9/25/2024 1:00:46 PM 127.0.0.1 Returned 304 in 8 ms
 HTTP  9/25/2024 1:00:46 PM 127.0.0.1 Returned 304 in 8 ms
 HTTP  9/25/2024 1:00:46 PM 127.0.0.1 Returned 304 in 7 ms
 HTTP  9/25/2024 1:00:46 PM 127.0.0.1 Returned 304 in 8 ms
 HTTP  9/25/2024 1:00:46 PM 127.0.0.1 GET /app.json
 HTTP  9/25/2024 1:00:46 PM 127.0.0.1 Returned 304 in 3 ms
2024-09-25 13:56:23,751 - Startup - startup     -  [INFO    ] - ==================== STARTING GivTCP==========================
Traceback (most recent call last):
  File "/app/startup.py", line 315, in <module>
    mask=str(interface['ipv4']['address']).split('/')[1][:-2]
         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^
IndexError: list index out of range
2024-09-25 13:58:42,216 - Startup - startup     -  [INFO    ] - ==================== STARTING GivTCP==========================
Traceback (most recent call last):
  File "/app/startup.py", line 316, in <module>
    mask=str(interface['ipv4']['address']).split('/')[1][:-2]
         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^
IndexError: list index out of range
2024-09-25 14:00:01,240 - Startup - startup     -  [INFO    ] - ==================== STARTING GivTCP==========================
Traceback (most recent call last):
  File "/app/startup.py", line 316, in <module>
    mask=str(interface['ipv4']['address']).split('/')[1][:-2]
         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^
IndexError: list index out of range
2024-09-25 14:04:34,448 - Startup - startup     -  [INFO    ] - ==================== STARTING GivTCP==========================
Traceback (most recent call last):
  File "/app/startup.py", line 316, in <module>
    mask=str(interface['ipv4']['address']).split('/')[1][:-2]
         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^
IndexError: list index out of range`

britkat1980, 2024-09-25:
Should be fixed in 3.0.3

>
> Alert!!!!
Just upgraded to 3.0.2 .... and it won't run!

mprowe, 2024-09-26:
Sorry. But....
Glad to see 3.0.3 this morning. However, I now seem to have lost connection with my battery?
Not sure how to pass the information in text-form, so it will have to be screenshot (sorry).
`
[image]
`