britkat1980/ha-addons #17: GIVTCP3 beta not starting: FileNotFoundError: [Errno 2] No such file or directory: 'v2env.pkl'

gibsta296, 2024-09-10:
I switch from v2.49 to v3 beta today, but it fails to load with the following error in the logs, it seems to find my Gen1 inverter, but fails with a file error.
Looks like v3 doesn't have a config section, so not sure how I can fix the error, any suggestions ?

2024-09-10 09:34:36,449 - Startup - startup     -  [INFO    ] - ==================== STARTING GivTCP==========================
2024-09-10 09:34:38,473 - Startup - startup     -  [INFO    ] - Searching for Inverters
2024-09-10 09:34:38,474 - Startup - startup     -  [INFO    ] - Scanning network for GivEnergy Devices...
2024-09-10 09:34:51,798 - Startup - client      -  [INFO    ] - Connection established to 10.0.0.6:8899
2024-09-10 09:34:51,798 - Startup - client      -  [INFO    ] - Detecting plant
2024-09-10 09:34:53,040 - Startup - client      -  [INFO    ] - Plant Detected
2024-09-10 09:34:55,815 - Startup - startup     -  [INFO    ] - Inverter SD2209G155 which is a Gen1 - Hybrid with 1 batteries has been found at: 10.0.0.6
2024-09-10 09:34:55,822 - Startup - startup     -  [INFO    ] - Inverter SD2209G155 not in settings file
2024-09-10 09:34:55,823 - Startup - startup     -  [INFO    ] - Adding Inverter SD2209G155 to slot 1
Traceback (most recent call last):
  File "/app/startup.py", line 449, in <module>
    with open('v2env.pkl','rb') as inp:
         ^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: 'v2env.pkl'

britkat1980, 2024-09-10:
What version gives you that error? I think I fixed that bug last night, but will need to check later today

gibsta296, 2024-09-10:
GivTCP-BETA
Current version: 3.0.0e

britkat1980, 2024-09-10:
Ah apologies, I didn't push the fix to beta, only dev... Try Beta3.0.0f

gibsta296, 2024-09-10:
It's getting further (I can complete the config sections), but it fails to start

2024-09-10 10:13:52,877 - Startup - startup     -  [INFO    ] - ==================== STARTING GivTCP==========================
2024-09-10 10:13:54,903 - Startup - startup     -  [INFO    ] - Searching for Inverters
2024-09-10 10:13:54,903 - Startup - startup     -  [INFO    ] - Scanning network for GivEnergy Devices...
2024-09-10 10:14:09,139 - Startup - client      -  [INFO    ] - Connection established to 10.0.0.6:8899
2024-09-10 10:14:09,139 - Startup - client      -  [INFO    ] - Detecting plant
2024-09-10 10:14:10,399 - Startup - client      -  [INFO    ] - Plant Detected
2024-09-10 10:14:13,158 - Startup - startup     -  [INFO    ] - Inverter SD2209G155 which is a Gen1 - Hybrid with 1 batteries has been found at: 10.0.0.6
2024-09-10 10:14:13,169 - Startup - startup     -  [INFO    ] - Inverter SD2209G155 already found in settings file (slot 1), checking IP address is unchanged...
2024-09-10 10:14:13,171 - Startup - startup     -  [INFO    ] - Setting up invertor: 1
2024-09-10 10:14:13,216 - Startup - startup     -  [INFO    ] - ==============================================================
2024-09-10 10:14:13,217 - Startup - startup     -  [INFO    ] - ====             Web Gui Config is at                     ====
2024-09-10 10:14:13,217 - Startup - startup     -  [INFO    ] - ====     http://10.0.0.28:8099/config.html             ====
2024-09-10 10:14:13,217 - Startup - startup     -  [INFO    ] - ==============================================================
2024-09-10 10:14:13,218 - Startup - startup     -  [INFO    ] - Running Invertor 1 (SD2209G155) read loop every 30/120s
2024-09-10 10:14:13,225 - Startup - startup     -  [INFO    ] - Serving Web Dashboard from port 3000
[2024-09-10 10:14:14 +0100] [77] [INFO] Starting gunicorn 23.0.0
[2024-09-10 10:14:14 +0100] [77] [INFO] Listening at: http://0.0.0.0:6345 (77)
[2024-09-10 10:14:14 +0100] [77] [INFO] Using worker: sync
[2024-09-10 10:14:14 +0100] [89] [INFO] Booting worker with pid: 89
[2024-09-10 10:14:14 +0100] [90] [INFO] Booting worker with pid: 90
[2024-09-10 10:14:14 +0100] [91] [INFO] Booting worker with pid: 91
2024-09-10 10:14:15,651 - Inv1 - read        -  [INFO    ] - Starting watch_plant loop...
2024-09-10 10:14:15,666 - Inv1 - read        -  [CRITICAL] - Detecting inverter characteristics...
 INFO  Accepting connections at http://localhost:3000
2024-09-10 10:15:07,053 - Inv1 - read        -  [ERROR   ] - Error processing Inverter data: ((<class 'TypeError'>, TypeError("argument of type 'Battery' is not iterable"), <traceback object at 0x7ffa39a06a00>), 723)
2024-09-10 10:15:07,056 - Inv1 - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7ffa39a06900>), 1572)
2024-09-10 10:15:07,056 - Inv1 - read        -  [ERROR   ] - runAll2 Error processing registers: ('KeyError', 'read.py', 1750)
2024-09-10 10:15:38,142 - Inv1 - read        -  [ERROR   ] - Error processing Inverter data: ((<class 'TypeError'>, TypeError("argument of type 'Battery' is not iterable"), <traceback object at 0x7ffa399f19c0>), 723)
2024-09-10 10:15:38,143 - Inv1 - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7ffa399debc0>), 1572)
2024-09-10 10:15:38,143 - Inv1 - read        -  [ERROR   ] - runAll2 Error processing registers: ('KeyError', 'read.py', 1750)

gibsta296, 2024-09-10:
I only have one inverter and one battery (which the config file seems to show correctly)

gibsta296, 2024-09-10:
That's with 
GivTCP-DEV
Current version: 3.0.0y

gibsta296, 2024-09-10:
I will try beta F

gibsta296, 2024-09-10:
Installed Beta F, getting the following in the logs.

024-09-10 10:33:22,276 - Startup - startup     -  [INFO    ] - ==================== STARTING GivTCP==========================
2024-09-10 10:33:24,303 - Startup - startup     -  [INFO    ] - Searching for Inverters
2024-09-10 10:33:24,303 - Startup - startup     -  [INFO    ] - Scanning network for GivEnergy Devices...
2024-09-10 10:33:37,964 - Startup - client      -  [INFO    ] - Connection established to 10.0.0.6:8899
2024-09-10 10:33:37,964 - Startup - client      -  [INFO    ] - Detecting plant
2024-09-10 10:33:39,205 - Startup - client      -  [INFO    ] - Plant Detected
2024-09-10 10:33:41,980 - Startup - startup     -  [INFO    ] - Inverter SD2209G155 which is a Gen1 - Hybrid with 1 batteries has been found at: 10.0.0.6
2024-09-10 10:33:42,016 - Startup - startup     -  [INFO    ] - Inverter SD2209G155 already found in settings file (slot 1), checking IP address is unchanged...
2024-09-10 10:33:42,022 - Startup - startup     -  [INFO    ] - Setting up invertor: 1
2024-09-10 10:33:42,062 - Startup - startup     -  [INFO    ] - ==============================================================
2024-09-10 10:33:42,063 - Startup - startup     -  [INFO    ] - ====             Web Gui Config is at                     ====
2024-09-10 10:33:42,063 - Startup - startup     -  [INFO    ] - ====     http://10.0.0.28:8099/config.html             ====
2024-09-10 10:33:42,063 - Startup - startup     -  [INFO    ] - ==============================================================
2024-09-10 10:33:42,064 - Startup - startup     -  [INFO    ] - Running Invertor 1 (SD2209G155) read loop every 30/120s
2024-09-10 10:33:42,067 - Startup - startup     -  [INFO    ] - Serving Web Dashboard from port 3000
[2024-09-10 10:33:42 +0100] [77] [INFO] Starting gunicorn 23.0.0
[2024-09-10 10:33:42 +0100] [77] [INFO] Listening at: http://0.0.0.0:6345 (77)
[2024-09-10 10:33:42 +0100] [77] [INFO] Using worker: sync
[2024-09-10 10:33:42 +0100] [89] [INFO] Booting worker with pid: 89
[2024-09-10 10:33:42 +0100] [90] [INFO] Booting worker with pid: 90
[2024-09-10 10:33:42 +0100] [91] [INFO] Booting worker with pid: 91
2024-09-10 10:33:44,862 - Inv1 - read        -  [INFO    ] - Starting watch_plant loop...
2024-09-10 10:33:44,871 - Inv1 - read        -  [CRITICAL] - Detecting inverter characteristics...
 INFO  Accepting connections at http://localhost:3000
2024-09-10 10:34:32,977 - Inv1 - read        -  [ERROR   ] - Error processing Inverter data: ((<class 'TypeError'>, TypeError("argument of type 'Battery' is not iterable"), <traceback object at 0x7fa128e33d00>), 723)
2024-09-10 10:34:32,978 - Inv1 - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7fa1291f0f80>), 1572)
2024-09-10 10:34:32,978 - Inv1 - read        -  [ERROR   ] - runAll2 Error processing registers: ('KeyError', 'read.py', 1750

gibsta296, 2024-09-13:
I tried rolling back to v2.49 but it no longer works

I tried the beta version: 3.0.0h, looking at the logs, it seems to start, but anytime I try to use the config page, it throws an error "Error: SyntaxError: Unexpected token '<', "<html> <h"... is not valid JSON", which means I can't configure GIVTCP (the MQTT option is disabled, I can't set it to enabled)

The dashboard page loads, but all values are zero

britkat1980, 2024-09-21:
Fixed in v3 release

gibsta296, 2024-10-01:
I tried installing latest GivTVP-DEV v3.0.12, which starts and finds invertor, but it doesn't show anything and I get an error when I try to read or save the config info

Error: SyntaxError: Unexpected token '<', "<html> <h"... is not valid JSON

I have also tried going back to v2, but I can't get anything to work at all now, any suggestions ?