britkat1980/ha-addons #19: 3.0 upgrade process

alfwro13, 2024-09-16:
In Home Assistant I have added the new repository as per the release notes on the 2.4.9 version. But I am unclear as what exactly is the upgrade process? Now that 3.0 is out do I stop the old add-on and start the new add-on? Will 3.0 keep the entity names the same?
Some more info would be much appreciated.
Thanks

SwiftRR, 2024-09-16:
I have just tried upgrading from v2.4.9 and am now successfully (I hope) running v3.0.0
I downloaded the new add on repo https://github.com/britkat1980/ha-addons which showed up as an add on along with beta and dev versions.
I installed v3.0.0
I then stopped v2.4.9 and started v3.0.0
_NOTE: I did first start v3.0.0 without stopping v2.4.9. This confused HA and I got loads of errors_

I checked givTCP and MQTT logs and everything seems to have been picked up automatically.

I am running Predbat which continued to run with fine. I have seen no errors.

My only hiccup has been with the 'configuration' tab. If I select this, I get a blank screen with the message:

> This add-on has no configuration.  

I can access the web-based configuration screen by either opening the URL from a new webpage or by selecting 'WebUI'
I found that the webUI had been populated with my existing values. I did disable the EV which seems to be added by default. 

v3.0.0 does seem to be running more quickly in the web portal and app (on IoS for me).

I have added givTCP to the sidebar. This takes me to this screen:
[image]

From here, Configuration and Readme both work.
But 'View Dashboard' takes me to this error page:
[image]

So several glitches. I now need to monitor how v3.0.0 runs.

Rob

britkat1980, 2024-09-16:
View Dashboard requires you to have enabled the web-dash in the config. If this is not enabled, then he dashboard will not be available.

The configuration tab in HA Addon will no longer have any items as they are now managed via the web-gui

gcoan, 2024-09-17:
Bit weird my own v3 install experience.

I already had the ha-addons folder in my app store from trying a beta version some time ago, so GivTCP v3, install.

After a while of the install executing I got an error message ‘install failed, see supervisor logs for detail’ and the addon had the ‘install’ option again, no ‘start’ option.

Checked the supervisor logs, no errors, all looked to have worked:

```
2024-09-17 09:11:24.167 INFO (MainThread) [supervisor.addons.addon] No 533ea71a_givtcp addon Docker image givtcp.docker.scarf.sh/britkat/giv_tcp-ma found
2024-09-17 09:11:24.168 INFO (MainThread) [supervisor.docker.interface] Downloading docker image givtcp.docker.scarf.sh/britkat/giv_tcp-ma with tag 3.0.0.
2024-09-17 09:12:29.496 INFO (MainThread) [supervisor.addons.addon] Creating Home Assistant add-on data folder /data/addons/data/533ea71a_givtcp
2024-09-17 09:12:29.504 INFO (MainThread) [supervisor.docker.interface] Downloading docker image givtcp.docker.scarf.sh/britkat/giv_tcp-ma with tag 3.0.0.
2024-09-17 09:12:31.097 INFO (MainThread) [supervisor.addons.manager] Add-on '533ea71a_givtcp' successfully installed
```

Went back to the GivTCP v3 add-on and now it had ‘start on boot’ ticked and the option to start the addon.  No idea why initially HA thought the install had failed when it doesn’t appear to have

britkat1980, 2024-09-21:
> My only hiccup has been with the 'configuration' tab. If I select this, I get a blank screen with the message:
> 
> > This add-on has no configuration.
> 
> I can access the web-based configuration screen by either opening the URL from a new webpage or by selecting 'WebUI' I found that the webUI had been populated with my existing values. I did disable the EV which seems to be added by default.
> 
> v3.0.0 does seem to be running more quickly in the web portal and app (on IoS for me).
> 
> 
> From here, Configuration and Readme both work. But 'View Dashboard' takes me to this error page: 

Configuration is now entirely via the web gui

Web dashboard must be enabled in the config before it will display

closed 2024-09-21 by britkat1980