britkat1980/ha-addons #20: Spikes and dips to zero in load energy today

naltsta, 2024-09-16:
[image]

2024-09-16 17:55:05,496 - Inv1 - write       -  [INFO    ] - Setting Eco mode was a success
2024-09-16 17:55:06,071 - Inv1 - write       -  [INFO    ] - Setting battery charge rate 2610 was a success
2024-09-16 17:55:06,721 - Inv1 - write       -  [INFO    ] - Setting battery discharge limit 2610 was a success
2024-09-16 17:55:23,386 - Startup - startup     -  [ERROR   ] - Error in watchdog loop: ('NameError', 'startup.py', 688)
2024-09-16 17:56:23,387 - Startup - startup     -  [ERROR   ] - Error in watchdog loop: ('NameError', 'startup.py', 688)
2024-09-16 17:57:23,388 - Startup - startup     -  [ERROR   ] - Error in watchdog loop: ('NameError', 'startup.py', 688)
2024-09-16 17:58:23,389 - Startup - startup     -  [ERROR   ] - Error in watchdog loop: ('NameError', 'startup.py', 688)
2024-09-16 17:59:23,391 - Startup - startup     -  [ERROR   ] - Error in watchdog loop: ('NameError', 'startup.py', 688)
2024-09-16 18:00:23,392 - Startup - startup     -  [ERROR   ] - Error in watchdog loop: ('NameError', 'startup.py', 688)
2024-09-16 18:01:23,393 - Startup - startup     -  [ERROR   ] - Error in watchdog loop: ('NameError', 'startup.py', 688)
2024-09-16 18:02:23,394 - Startup - startup     -  [ERROR   ] - Error in watchdog loop: ('NameError', 'startup.py', 688)
2024-09-16 18:03:23,395 - Startup - startup     -  [ERROR   ] - Error in watchdog loop: ('NameError', 'startup.py', 688)
2024-09-16 18:04:23,397 - Startup - startup     -  [ERROR   ] - Error in watchdog loop: ('NameError', 'startup.py', 688)
2024-09-16 18:05:23,398 - Startup - startup     -  [ERROR   ] - Error in watchdog loop: ('NameError', 'startup.py', 688)
2024-09-16 18:06:23,399 - Startup - startup     -  [ERROR   ] - Error in watchdog loop: ('NameError', 'startup.py', 688)
2024-09-16 18:07:23,401 - Startup - startup     -  [ERROR   ] - Error in watchdog loop: ('NameError', 'startup.py', 688)
2024-09-16 18:08:23,402 - Startup - startup     -  [ERROR   ] - Error in watchdog loop: ('NameError', 'startup.py', 688)
2024-09-16 18:09:23,404 - Startup - startup     -  [ERROR   ] - Error in watchdog loop: ('NameError', 'startup.py', 688)
2024-09-16 18:10:23,405 - Startup - startup     -  [ERROR   ] - Error in watchdog loop: ('NameError', 'startup.py', 688)
2024-09-16 18:11:23,406 - Startup - startup     -  [ERROR   ] - Error in watchdog loop: ('NameError', 'startup.py', 688)

naltsta, 2024-09-16:
Turning off the car charger seems to have got rid of the error messages but I'm still getting the dips to zero

britkat1980, 2024-09-16:
The dips to zero are under investigation. A temp fix will come tomorrow, provided it proves effective overnight on teat

britkat1980, 2024-09-21:
Please test beta3.0.1 and report back any improvement (or not!)

naltsta, 2024-09-29:
Tried 3.0.3 today having stayed on version 2.

Installed just before 11am and saw a small spike.

No drops to zero but just had a massive spike[image]

IWR69, 2024-10-21:
I am seeing something similar on 2.4.9 but only when charging or discharging the battery.
[image]
(Charging at 2am to 5am)

As you can see in the image below the issue only seems to happen when using a charge or discharge schedule and doesn't seem to be affected by Eco charging during the day or discharging in the evening.
[image]
(First 2 highlighted areas are charges and the 4pm to 7pm is a discharge schedule)

I think this fluctuating load total is confusing my Utility Meter Helper which is setup to monitor the 4 tariffs (day, night, off peak and peak) and the peak and off peak helpers, which correspond to the highlighted areas in the image above, are showing values that are about 8 times what I expect.

I have a GE 5kWh Hybrid Inverter G3 (D0.311-A0.311 3018) and 9.5kWh Battery G2. I'm running HA 2024.10.3

IanJS46, 2025-05-25:
Wasn't sure if to add comments to this old case or start a new one..
I am running GivTCP V 3.1.6 to a GE Hybrid Gen 3  5kW Inverter with 2 x 9.5 kWh Batteries and I'm seeing the odd dip to zero on SOC and Temperature values from the batteries.  I didn't get this on GivTCP V2.x, Inverter is D0.316-A0.316, batteries both V 3020, I have not noticed this on any other analogue value but that may be because I don't monitor them.  My inverter is on a hardwired network.

[image]
[image]

Ian

gcoan, 2025-05-25:
> and I'm seeing the odd dip to zero on SOC and Temperature values from the batteries

If you look in the givtcp log at the times the dips occur do you see anything logged ?

IanJS46, 2025-05-26:
This morning caught this....

No errors in log, data only being written every 30 Seconds...

[image]

 
[image]

IanJS46, 2025-05-26:
Yes RTC is on

[image]

IanJS46, 2025-05-27:
Battery 069 temperature is now zero, I noticed all the 4 Cells were reporting zero temperature. Nothing in the log (It's now just gone 5pm)

[image]

[image]

IanJS46, 2025-06-13:
Hi what's the latest with this?  Is this a GivEnergy or GivTCP issue?  As you can see from the images below its still a problem.

[image]

[image]

Is there anything I can do my end?

Ian

IWR69, 2025-06-15:
This bug is proving to be a right pita and is causing me issues with my automations.
I hope this gets fixed soon.

[image]