britkat1980/ha-addons #21: Constant TypeErrors when reading from 5kW Gen3 inverter: `TypeError('list indices must be integers or slices, not str'`

peter1rhodes, 2024-09-18:
I've just upgraded to v3.0.0 from v2.4.9, and it is working well, except for my GivEnergy solar inverter, which is not being read correctly, and is giving constant errors in the GivTCP logs:

```
2024-09-18 14:13:18,703 - Inv1 - read        -  [ERROR   ] - Error processing Inverter data: ((<class 'TypeError'>, TypeError('list indices must be integers or slices, not str'), <traceback object at 0x7f9233afc0>), 871)
2024-09-18 14:13:18,704 - Inv1 - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7f92320a00>), 1587)
2024-09-18 14:13:18,705 - Inv1 - read        -  [ERROR   ] - runAll2 Error processing registers: ('KeyError', 'read.py', 1766)
```

peter1rhodes, 2024-09-18:
It may be useful context that this GivEnergy hybrid inverter does not have any batteries attached (I have a separate AIO)

peter1rhodes, 2024-09-18:
I believe the problem is in `read.py`, line 871:

```
        elif GEInv.battery_percent == 0 and 'multi_output_old' in locals():  # line 870
            power_output['SOC'] = multi_output_old['Power']['Power']['SOC']  # line 871
```

However, under some circumstances, `multi_output_old` is instantiated as a list (if not regCacheStack on lines 718-722), meaning this string index will give the error shown. My guess is that this section is written with the assumption that a hybrid inverter will always have a battery connected, so a state of 0 soc indicates some other issue or state? But in reality, this may not be the case.

I don't know where the code lives for this now, as the versions I found in both this repository (`ha_addons`) and your other repository `GivTCP` don't have these updates. Instead, I had to pull the docker image this repo refers to and rip the code out of that to inspect

britkat1980, 2024-09-18:
Thanks, the code is under the dev3 branch of the main repo (problems pushing it to main).

I think that the challenge is that in a scenario where there is an empty cache (multi_output_old=[] ), which should only be at startup, and the SOC is reported as Zero (usually an error, but in your scenario is valid as no batteries), GivTCP is still looking for the SOC value. I've tweaked code for this and will push in the next point release.

britkat1980, 2024-09-21:
out now in beta 3.0.1, can you test?

peter1rhodes, 2024-09-21:
Lovely, will do, thank you

peter1rhodes, 2024-09-21:
@britkat1980 Apologies for the slight off-topic; with the new configuration portal, is there a way to easily save/back up/transfer configs?

peter1rhodes, 2024-09-21:
@britkat1980 Yes, this seems to be working perfectly! I have comms back with my hybrid inverter, and can't see any errors in the log. Thank you very much

closed 2024-09-21 by britkat1980