britkat1980/ha-addons #23: v3 error with duplicate sensor id's for battery entities

gcoan, 2024-09-19:
I have two Gen 1 hybrid inverters, called G and H in GivTCP.

G has a single 9.5kW battery connected, H has a single 5.2 connected.

Since ugrading to GivTCP v3.0.0 I can see a whole load of errors in the HA log about duplicate id's from MQTT:

```
2024-09-19 21:45:14.822 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID G_DX2327G139_Battery_Capacity is already used by sensor.g_dx2327g139_battery_capacity - ignoring sensor.g_dx2327g139_battery_capacity_2
2024-09-19 21:45:14.824 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID G_DX2327G139_Battery_Cell_10_Voltage is already used by sensor.g_dx2327g139_battery_cell_10_voltage - ignoring sensor.g_dx2327g139_battery_cell_10_voltage_2
```

This is occurring for both batteries, but only for the battery entities, no such warnings from any of the inverter entities.

Shutdown HA and all the add-ons, and the errors re-appeared in the clean log.

Curiously looking at the voltage history, it looks like the barry sensors were not being read for the last day (I upgraded to v3 last night), but are now being read OK:
[image]

No such similar pattern with the inverter entities, e.g. SoC tracked OK overnight and through today.

Shutdown and restarted everything again in case it was a one-off set of errors, but no, they all still appear again my logfile.

gcoan, 2024-09-20:
Another observation, with MQTT Explorer I can see what appears to be duplicated battery sensors under the homeassistant/givenergy tree:

[image]

There are sensors named as <battery id>_voltage and also <inverter id>_<battery id>_voltage. Maybe one set created by GivTCP v2.4.9 and one set by v3 which is what's causing the problem?

SwiftRR, 2024-09-20:
I am having similar issues after moving to v3 from v2.4.9

My battery details are all 'unknown'.
I have 11 MQTT devices shown and I can see duplicates:
[image]

I have GivTCP DY2218G702 (all blank), followed by GivTCP DY2218G702 Battery details (all unknown).
With the same for GivTCP DY2219G082.

I can also see this error in System logs
[image]

I also see this error for my single inverter (I have 2 batteries)
[image]

Like @gcoan,  in MQTT Explorer all the inverter and battery info shows up twice:

1. under Givenergy
2. under homeassistant.
[image]

I am seeing one set of entries under 'Battery stack', the other **not** under 'Battery stack'
The HA data shows all the battery details under my inverter but twice. e.g.
[image]

And again
[image]

The inverter/Battery entries appear to be under 'Battery stack'.

In conclusion, the battery details do appear to be there. It is just that they cannot be directly views in Devices and Services/Devices (MQTT list)

Rob

britkat1980, 2024-09-20:
V3 is supposed to remove all old discovery messages, and replace with new, but it is clear that battery entries are not being removed. Temp fix is to manually delete the messages out of the MQTT broker, then restart GivTCP. This will remove the _2 entities and ensure your original entities are properly updated

gcoan, 2024-09-20:
> V3 is supposed to remove all old discovery messages, and replace with new, but it is clear that battery entries are not being removed. Temp fix is to manually delete the messages out of the MQTT [...]

@britkat1980 Which entries are to be removed from MQTT?

I assume its under homeassistant tree, is it the entities with just battery id prefix or inverter then battery id prefix, or both?

At the moment I have reverted back to 2.4.9 which is working fine

qamar-yaunos, 2024-09-20:
@britkat1980 I hope you're well.

I am also having the same issue but I don't have 2 batteries.

My install was done in January but I needed a replacement battery and the entities for the old battery are still showing.

Can the entities for the old battery just be deleted?

gcoan, 2024-09-20:
> My install was done in January but I needed a replacement battery and the entities for the old battery are still showing.
> 
> Can the entities for the old battery just be deleted?

I had my battery replaced earlier this year and had stale MQTT sensors for the old battery in HA.  You can't delete them directly from Home Assistant, need to install MQTT Explorer add-on and delete the messages from the MQTT queues

britkat1980, 2024-09-21:
Temp fix is to remove the items under the homeassistant topic which start with the battery serial number (rather than the inverter serial number).
3.0.1 should fix this, but only if you're upgrading. To force this you can remove (or rename) your config file and restart GivTCP

SwiftRR, 2024-09-21:
This morning, I installed v3 again from v2.4.9.
I could see all the Battery details. But if I then restart HA, I am back to square 1. So something in HA seems to be conflicting.

> Temp fix is to remove the items under the homeassistant topic which start with the battery serial number (rather than the inverter serial number).

Following your message above, I have removed the homeassistant topics starting with the battery serial number using MQTT Explorer. Tedious as I had to remove each line at a time, but now done.
That seems to have cleared up the 'unknown' issue.
Rebooting HA also didn't interfere.

I will see how v3 runs now.

> 3.0.1 should fix this, but only if you're upgrading. To force this you can remove (or rename) your config file and restart GivTCP

I can't see v3.0.1 and can't see how I could remove a config file as there is no longer a config file with the WebUI.

Rob

britkat1980, 2024-09-21:
Have you disabled auto-start switch on the v2 addon? Is it booting up and conflicting?

FYI, MQTT Explorer you can delete category topics, which saves individually deleting each topic!

SwiftRR, 2024-09-21:
Yes, I had the v2.4.9 auto-start disabled this morning.
Auto-start was only enabled for v3.
And the HA restart brought back the 'unknown' issue.

But clearing out the battery topics with MQTT Explorer seems to have done the trick and all the battery details stay. I am certainly testing this .......
Before I cleared the battery topics, Devices and Services shown 10 MQTT devices.
Afterwards, I had 9 MQTT devices showing. 

Rob

SwiftRR, 2024-09-21:
Sorry, v3 isn't working for me. I am reverting back to v2.4.9
See part of my add on log below. This has been repeating for he last 2 hours.
[image]

Rob

britkat1980, 2024-09-21:
So it looks like GivTCP got stuck and restarted the read loop. Does that keep looing round? There's only one restart here, but I assume there's more?

SwiftRR, 2024-09-21:
The log had been stuck at this point for 2 hours.
I tried to go back to v2.4.9 but I was reverted back to v2.4.3 with no option to upgrade to v2.4.9
I looked at the add on repositories and this is what was displayed for V3:
[image]
So something is really confused. I deleted v2 and am now running v3.
This shows battery details OK in MQTT Devices

So I am stuck on v3, I guess!
Hope for an update soon.

Rob

britkat1980, 2024-09-21:
There shouldn't necessarily be any more log line after that, the key question is were you not seeing data being updated in HA/MQTT?

SwiftRR, 2024-09-21:
I have updated to v3.0.1 beta.
I have checked MQTT/devices battery details which are all shown. 
I have restarted HA and all the MQTT devices seem to be unaffected.
I run predbat which is running smoothly with clean logs. System log also looks good.

However it look like v3.0.1 is restarting from time to time and there are some errors after which v3.01 starts running OK again. This is an extract from earlier this evening. I will see what it all looks like tomorrow.
[image]

When I tried to downgrade back to v8.4.9, I found myself in v2.4.3! So has v2.4.9 been withdrawn?

Rob

gcoan, 2025-12-02:
all resolved now