britkat1980/ha-addons #25: FR: Ability to not configure the tariff rates within GivTCP

gcoan, 2024-09-19:
Most of the features on GivTCP are configurable on or off, which is great, but not the tariff rates.

Unless you are on a simple standard rate tariff or Economy 7, these sensors are pretty useless.

In trying to reduce the size of my HA database I found that I had a few thousand unnecessary records in my state table from these sensors.

I subsequently purged

sensor.givtcp_xxxx_day_cost 
sensor.givtcp_xxxx_day_energy_kwh
sensor.givtcp_xxxx_day_energy_total
sensor.givtcp_xxxx_day_rate
sensor.givtcp_xxxx_day_start_energy
(and similar _night_ sensors)

and disabled the sensors in HA, but it would be better if GivTCP wasn't creating them and filling them with artificial (and wrong) cost rates & accumulations.

britkat1980, 2024-09-21:
Thanks, I'll look at this to see if its worth making configurable. I have Octopus Intelligent and I use the dynamic tariff mode to make the rate work. Use the dynamic tariff switch to track the rate in HA.

gcoan, 2024-09-21:
Thanks.  I am on Octopus Agile, have an Octopus mini and use the Octopus integration to track current accumulative cost, so haven't had to investigate the dynamic tariff option in givtcp.  As ever, multiple ways to achieve the same thing in HA