britkat1980/ha-addons #26: Strange v3.0.1 addon log: Inverter energy jumped too far

SwiftRR, 2024-09-22:
I installed givTCP 3.0.1 last night (Sat 22) and my addon log looks very strange. See extract below. 
Running on HAOS on pi.

Is my system still operating OK?

```
2024-09-22 06:55:19,773 - Inv1 - read        -  [INFO    ] - Invertor_Energy_Today_kWh jumped too far in a single read: 0.1->0.7 so using previous value
2024-09-22 06:55:36,065 - Inv1 - read        -  [INFO    ] - Invertor_Energy_Today_kWh jumped too far in a single read: 0.1->0.7 so using previous value
2024-09-22 06:55:39,767 - Inv1 - write       -  [INFO    ] - Setting Charge Slot 1 to: 07:00 - 09:00 was a success
2024-09-22 06:55:40,519 - Inv1 - write       -  [INFO    ] - Setting Charge Schedule to enable was a success
2024-09-22 06:55:40,994 - Inv1 - transparent -  [CRITICAL] - Function code 86 recieved. Gracefully handled
2024-09-22 06:55:41,278 - Inv1 - write       -  [INFO    ] - Setting Battery Pause Mode to Disabled was a success
2024-09-22 06:55:51,400 - Inv1 - write       -  [INFO    ] - Setting battery reserve 4 was a success
2024-09-22 06:55:52,920 - Inv1 - read        -  [INFO    ] - Invertor_Energy_Today_kWh jumped too far in a single read: 0.1->0.7 so using previous value
2024-09-22 06:56:09,464 - Inv1 - read        -  [INFO    ] - Invertor_Energy_Today_kWh jumped too far in a single read: 0.1->0.7 so using previous value
2024-09-22 06:56:26,008 - Inv1 - read        -  [INFO    ] - Invertor_Energy_Today_kWh jumped too far in a single read: 0.1->0.7 so using previous value
2024-09-22 06:56:42,243 - Inv1 - read        -  [INFO    ] - Invertor_Energy_Today_kWh jumped too far in a single read: 0.1->0.7 so using previous value
2024-09-22 06:56:58,793 - Inv1 - read        -  [INFO    ] - Invertor_Energy_Today_kWh jumped too far in a single read: 0.1->0.7 so using previous value
```
I run Predbat which is operating as expected with a clean addon log during same time as v3.0.1 log above:

```
2024-09-22 06:50:11.443450: Info: record_status Hold charging target 91%-91%
Trigger callback for select select_option
2024-09-22 06:55:52.981300: Info: record_status Idle
2024-09-22 07:00:12.498120: Info: record_status Charging target 91%-100%
```

Rob

peter1rhodes, 2024-09-22:
I am experiencing the same issue. I also use predbat, should that be relevant. The logs are full of this message (`Invertor_Energy_Today_kWh jumpted too far in a single read...`). This is being logged for both my AIO and my hybrid inverter. When this first started, the 'large jump' was 0.1->0.2, but over the day it's gotten steadily larger (0.1->0.3 for a while, then 0.1->0.4, etc etc)

SwiftRR, 2024-09-22:
I have noticed that this suspected 'bug' coincides when the inverter is idle.
When there is something active to do, such as charging or discharging, the log settles down to expected behaviour.

I also sometimes see this error, which seems to clear after it restarts.
`2024-09-22 13:26:37,956 - Startup - startup     -  [ERROR   ] - Self Run loop process stuck. Killing and restarting..`.

I did restart v3.0.1 late morning and the log has looked sound since.
I will continue monitoring its behaviour.

Rob