britkat1980/ha-addons #27: GivEVC prefixes

peter1rhodes, 2024-09-22:
In GivTCP, we can add prefixes to our devices, to make them easier to search or identify within HA. THis is fine, but it appears to apply unpredictably to the EV charger. In my experience, the EV charger will be given one of the inverter prefixes, but it's not always obvious which, and I'm sure it seems to change sometimes!

It should be that just like other "inverters" (whether gateways, AIOs, or hybrid inverters), the EVC should have a configurable optional prefix.

peter1rhodes, 2024-09-30:
@britkat1980 I would be interested in contributing to the project. Is there a way I can help, or get involved? Apologies for sending this as a comment, I don't have another way to reach you.

britkat1980, 2024-09-30:
If you wanna pull a fork of GivTCp (dev3 branch) and then create a PR to cover the configurable EVC prefix that'd be great

peter1rhodes, 2024-09-30:
OK I will give that a go when I get chance. I may need your help in getting an environment set up and better understanding the code!

On another note, do you have anything from GivEnergy regarding the firmware bug that prevents modbus over ethernet from working reliably? I find it works for me sometimes but suddenly drops out and won't recover. Your README on the GivTCP repo says "_NOTE - there is currently a firmware bug with the GivEVC where the Modbus port only available on WiFi. Therefore if your EVC is connected via ethernet is will be necessary to disconnect and re-connect via WiFi to recieve data via GivTCP_." but when I've asked GivEnergy about this bug they claim no knowledge and say it's not their problem 😣

LeeFarm11, 2024-09-30:
That is interesting. I have connected Inverter and EV Charger via Ethernet cables with fixed IP. GivTCP has been given the fixed Ethernet IP addresses. Do I have a problem waiting to bite me?

peter1rhodes, 2024-09-30:
That's the setup I had, and I had issues with the EV charger (not the inverter, particularly). I'm just one person, but let me put it this way: if your setup is still working in a month, please let me know! I will grill you for every detail of your setup haha

LeeFarm11, 2024-09-30:
Unfortunatley I don't have an EV, only the EV Charger. Therefore I won't be able to see any functional issues. But I guess I would see some communication issues if I went looking for them? What would I see if I do have the problem?

britkat1980, 2024-09-30:
> @LeeFarm11 (https://github.com/LeeFarm11)  That's the setup I had, and I had issues with the EV charger (not the inverter, particularly). I'm just one person, but let me put it this way: if your setup is still working in a month, please let me know! I will grill you for every detail of your setup haha

EVC won’t connect via Ethernet as far as we know. So that might be why @Lee you are seeing EVC errors

peter1rhodes, 2024-09-30:
@britkat1980 if I understand correctly, @LeeFarm11 currently _isn't_ seeing issues. I myself have definitely had the EV charger working via ethernet, so it definitely _can_ work, but I find it unreliable — after some hours or days the connection fails and does not resume. That said, it's also unreliable over WiFi, so I haven't been able to define the problem as definitely related to Ethernet.

@LeeFarm11 despite you not having an EV, in my experience you will still see the issue if you routinely check on the EV charger device in HA, or on the GivTCP logs. The logs will report a refused or reset connection, and the device in HA will stop updating its sensors (including ones that don't rely on a car charging, like the "system time").

LeeFarm11, 2024-09-30:
I just took a look at the GivTCP log file. I searched for evc and I got no results. (since midnight)
I restarted GivTCP and I see plenty of log file records since restart relating to evc. So I guess the evc stopped responding and then GivTCP gave up trying to connect?

I see this shortly after restart-
2024-09-30 13:20:48,910 - evc - [DEBUG] - Error: ('ConnectionException', 'evc.py', 123)

And then it settles down to this every 10 seconds -

```
2024-09-30 13:21:08,981 - evc - [DEBUG] - Socket is already open
2024-09-30 13:21:09,014 - evc - [DEBUG] - Socket is still open
2024-09-30 13:21:09,026 - evc - [DEBUG] - Dealt with Charger
2024-09-30 13:21:09,026 - evc - [DEBUG] - Publish all to MQTT
2024-09-30 13:21:09,026 - evc - [DEBUG] - Publishing: GivEnergy/11288853538258/Charger
2024-09-30 13:21:19,031 - evc - [DEBUG] - Socket is already open
2024-09-30 13:21:19,036 - evc - [DEBUG] - Socket is still open
2024-09-30 13:21:19,049 - evc - [DEBUG] - Dealt with Charger
2024-09-30 13:21:19,049 - evc - [DEBUG] - Publish all to MQTT
2024-09-30 13:21:19,050 - evc - [DEBUG] - Publishing: GivEnergy/11288853538258/Charger
2024-09-30 13:21:29,065 - evc - [DEBUG] - Socket is already open
2024-09-30 13:21:29,068 - evc - [DEBUG] - Socket is still open
2024-09-30 13:21:29,083 - evc - [DEBUG] - Dealt with Charger
2024-09-30 13:21:29,083 - evc - [DEBUG] - Publish all to MQTT
2024-09-30 13:21:29,084 - evc - [DEBUG] - Publishing: GivEnergy/11288853538258/Charger
```

peter1rhodes, 2024-09-30:
Yeah `ConnectionException` sounds likely to be the issue I encountered. What GivTCP version are you running?

This conversation has moved away from the posted title a bit (this is partly my fault as I originally piggy-backed it to ask britkat a question. @britkat1980 where would be best for a conversation about individuals' experiences with EVC connectivity, dropouts etc? From what you said, it was perhaps surprising to you that EVC control over ethernet worked at all, so maybe there's something useful to be learned?