britkat1980/ha-addons #28: v3 settings blocked via CORS

jmhunter, 2024-10-04:
Hi

I have just upgraded from v2 to v3.

On my HA instance (which I access via its internal DNS name, `my-ha-01`), I go to `http://my-ha-01:8123/hassio/addon/533ea71a_givtcp/info` and from there I can 'Open Web UI' which takes me to `http://my-ha-01:8099/config.html`

However the config doesn't seem to work and in the Chrome browser console I can see the following message:
`Access to fetch at 'http://192.168.1.10:8099/REST1/settings' from origin 'http://my-ha-01:8099' has been blocked by CORS policy: No 'Access-Control-Allow-Origin' header is present on the requested resource. If an opaque response serves your needs, set the request's mode to 'no-cors' to fetch the resource with CORS disabled.`

In the HA add-on logs, I can see a number of `Error in watchdog loop: ('NameError', 'startup.py', 693)` errors which I presume is down to the setup/configuration not working; plus one `EVC IP is missing from config. Please update and restart GivTCP` (possibly the same)

I note that the add-on logs also say `v2 settings found, importing into v3...` but this doesn't seem to have worked. (The automatic plant detection in v3 did find a random modbus EE11a I have on the same subnet as my HA server but it didn't pick up my GE inverters which are on a different subnet)

jmhunter, 2025-08-20:
I have waited until now to try this again, as I hadn't seen a reply to this but figured I might as well give it a go.

I'm happy to confirm that with GivTCP v3.1.6 I can now navigate to the config web page and give it the IP addresses of my inverters - all good with that now.

(It still tries to autodiscover modbus adapters on the same subnet, and tries to query those as if they had givenergy inverters on them, but luckily none of my other modbus kit seems to have been impacted by that so I'm going to ignore that for now)

closed 2025-08-20 by jmhunter

gcoan, 2025-08-21:
> (It still tries to autodiscover modbus adapters on the same subnet, and tries to query those as if they had givenergy inverters on them, but luckily none of my other modbus kit seems to have been impacted by that so I'm going to ignore that for now)

if you edit the allsettings.json file in /config/GivTCP you can turn the autoscan off

jmhunter, 2025-08-21:
> if you edit the allsettings.json file in /config/GivTCP you can turn the autoscan off

nice - thank you. That wasn't in my allsettings.json at all, but from https://github.com/britkat1980/giv_tcp/blob/main/settings.json I saw that adding `"auto_scan":false` would do the trick - that worked.

jmhunter, 2025-08-21:
> It was from memory so sorry for getting the json file name wrong. Glad you found the right setting

The filename was correct, sorry I wasn't clear :) I just found the setting details in the above file (it wasn't in the allsettings.json on my instance, so I went hunting for a 'defaults' type file in the repo so I could check what the syntax should be)