britkat1980/ha-addons #31: Error with sync command

sjs-1956, 2024-10-21:
Ac coupled inverter Gen 1 - set as not old firmware
GIVTCP v2.4.9 - possibly earlier

Noticed the following the log. Not sure how long this has been happening. Over the last 24 hours not one sync command has succeeded

2024-10-21 07:03:52,989 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.1.62, 8899) failed: timed out
2024-10-21 07:07:55,203 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.1.62, 8899) failed: timed out
2024-10-21 07:07:57,708 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.1.62, 8899) failed: timed out
2024-10-21 07:08:00,214 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.1.62, 8899) failed: timed out
2024-10-21 07:08:02,719 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.1.62, 8899) failed: timed out

gcoan, 2024-10-21:
These could just be comms or inverter 'not talking' issues if they have just started happening.

Try shutting down and restarting your inverter and doing a 'reset to defaults' in the givenergy portal to reset the inverter configuration.  I've seen either of these fixes cure connection errors in givtcp

sjs-1956, 2024-10-29:
Thanks for your advice. It was confusing as the inverter was communicating locally with this app but not to GivEnergy
 cloud. Once I sorted that (new EE router problem) these log messages went away. Thanks again.

closed 2024-10-29 by sjs-1956