britkat1980/ha-addons #33: Sudden loss of data collection

steve-daedilus, 2024-11-24:
Overnight, my HA environment was no longer able to collect data. Only manual actions in leadup:

Proxmox VM migrated to different storage.

Actions taken following:
Restart of MQTT, GivTCP, Predbat and other services several times.
Restart of HA.
Restart of inverter.
Manual override attempts to start force charging via GivEnergy software controls.

No success on any.

Downgraded HA core auto update to previous version as per HA docs. No change.

Full powered off inverter, batteries. Then powered back on, and reset defaults for battery control states in the inverter settings.

Can now force charge batteries.

Still no data feeds. Really stuck on how to unpick this at this point, not sure if MQTT, GivTCP, HA or Predbat.

GivTCP logs:

[logs.txt](https://github.com/user-attachments/files/17893145/logs.txt)

gcoan, 2024-11-24:
It looks like your inverter is failing to talk to GivTCP.

Have you tried a full shutdown and restart of HA and all the add-on's, including MQTT.  It sounds like you have, which points to either a comms issue with the inverter not being on the network properly or the inverter being 'confused' which has happened to me before. 

Powering down, restarting the inverter and doing a reset to defaults are the usual things I try when I have this issue.  Could try changing the inverter mode from Eco to Force Charge or Discharge as well.

Is the inverter reporting correctly in the GE app and portal?

What version GivTCP

steve-daedilus, 2024-11-24:
Hi thanks for feedback.

GE Cloud data etc are reporting correctly. I did shutdown everything (shall go again anyway!)

britkat1980, 2024-11-24:
Can you delete all pkl files in the config folder and restart givtcp

gcoan, 2024-11-24:
I'm thinking this could be an inverter/modbus issue, which I've seen before, that the inverter talks to the cloud OK but wouldn't accept modbus connections from givtcp.

It might be something that needs givenergy support to do a different type of reset/restart of the inverter.

Some people have suggested that this is due to the HA upgrade.  Are you able to take a full backup of the current HA (core and all the add-on's) and then try restoring from an older HA backup to rule this in/out?

If you take a backup now you can then recover back to this backup when you've tried an old restore

steve-daedilus, 2024-11-24:
Hi both.

I always ensure backups are taken when upgrades done. So yes, I'm happy to backup and the roll back to a previous backup.

As for the pkl files - these are the ones located in homeassistant/GivTCP/ folder?

steve-daedilus, 2024-11-24:
Roll back to previous backup does appear to have solved the issue on face value. I'm monitoring now for a short while.

gcoan, 2024-11-24:
> As for the pkl files - these are the ones located in homeassistant/GivTCP/ folder?

yes, just delete them all and restart GivTCP