britkat1980/ha-addons #36: Unable to install v3.0.4

jon798, 2024-12-07:
Trying to install v3.0.4 (currently running v2.4.9), but when I hit 'install', I get the following error:

Can't install givtcp.docker.scarf.sh/britkat/giv_tcp-ma:3.0.4: 404 Client Error for http+docker://localhost/v1.47/images/givtcp.docker.scarf.sh/britkat/giv_tcp-ma:3.0.4/json: Not Found ("No such image: givtcp.docker.scarf.sh/britkat/giv_tcp-ma:3.0.4")

(Trying to install on HA v 2024.12.1 on HA OS 14.0 via the addon store).

rbroker, 2024-12-29:
Hi @jon798,

I had a similar issue trying to update from 3.0.0:
```
2024-12-29 15:51:32.366 INFO (MainThread) [supervisor.docker.addon] Updating image givtcp.docker.scarf.sh/britkat/giv_tcp-ma:3.0.0 to givtcp.docker.scarf.sh/britkat/giv_tcp-ma:3.0.4
2024-12-29 15:51:32.367 INFO (MainThread) [supervisor.docker.interface] Downloading docker image givtcp.docker.scarf.sh/britkat/giv_tcp-ma with tag 3.0.4.
2024-12-29 15:51:33.972 ERROR (MainThread) [supervisor.docker.interface] Can't install givtcp.docker.scarf.sh/britkat/giv_tcp-ma:3.0.4: 404 Client Error for http+docker://localhost/v1.47/images/givtcp.docker.scarf.sh/britkat/giv_tcp-ma:3.0.4/json: Not Found ("No such image: givtcp.docker.scarf.sh/britkat/giv_tcp-ma:3.0.4")
```
I'm not a docker expert by any means, but I think I was able to work around it by explicitly pulling the docker image of the appropriate version from the command line (I'm running on a Raspberry Pi Linux instance):
> docker image pull givtcp.docker.scarf.sh/britkat/giv_tcp-ma:3.0.4
It looks like something in the update process might have thought it had the correct container image downloaded, but it hadn't worked, so it was getting itself confused.

jon798, 2025-01-11:
Thanks @rbroker, but I'm running HassOS on Raspbery Pi and the docker command isn't there. I'm reluctant to start trying to install docker etc. on HassOS in case I mess anything else up...

Any other things I could try?

Cheers.

jon798, 2025-01-20:
In case it's useful to someone else: I updated my HA instance from a 32bit original build to a current 64bit core install, took a full backup beforehand and restored it to the rebuilt HassOS/Raspberry Pi and then was able to install v3.0.4 as an HA addon.

No idea why...

closed 2025-01-24 by jon798