britkat1980/ha-addons #37: FR: Add inverter writes counter

gcoan, 2024-12-09:
There has been quite a bit of "discussion" in the last week about the flash memory of GivEnergy inverters and how they can fail with high numbers of inverter writes.

@springfall2008 has made changes to Predbat to reduce the number of inverter writes and has added a total inverter writes counter, but this is of course only Predbat's idea of what an inverter write is.  It has no knowledge of whether a REST or HA sensor change results in 1, 2 or more actual modbus register writes on the GivEnergy inverter, and of course it only provides this data if you are using Predbat.  Some people are using their own scripts.

It would be very useful (and more accurate) if GivTCP could count the actual inverter register writes so we have a single version of truth.

Probably just an accumulated total per inverter and people can make their own daily/weekly totals with utility meters if they want to.

britkat1980, 2024-12-09:
I’ve literally just put this in dev 3.0.47 so will be available, will increment a counter with each register that is written.

gcoan, 2024-12-09:
> I’ve literally just put this in dev 3.0.47 so will be available, will increment a counter with each register that is written.

Ha, brilliant.  Thanks

gcoan, 2025-04-06:
Finally got a chance to see the new inverter writes feature as I've upgraded to beta 3.1.2

and @britkat1980  its not as useful as I thought it would be as the inverter write count resets to zero when the givtcp add-on is restarted:

[image]

the write count needs to be held more permanently so it survives restarts of the addon, e.g. in the json or another config file

closed 2025-12-02 by gcoan