britkat1980/ha-addons #41: v3.0.4 Entire previous day's figures added at midnight to energy dashboard

cbrinks17, 2025-01-02:
On the 31st when I encountered the error that many people did with the inverter time being incorrect. I was on version 2.8.4 at the time, and I thought something was wrong with my Giv TCP install, so I installed version 3. After I got it up and running, I realized that I needed to press the Send Time on the GE cloud. This understandably generated a large dump of the previous day's data when the issue was resolved. However, I have noticed that this also has happened on the 1st and 2nd of January.

31st December
[image]

1st January
[image]

2nd January
[image]

I am using the entity 'sensor.givtcp_{Serial}_load_energy_today_kwh' to track my house load in the dashboard. I know that this resets every day at midnight back to 0, and the history does seem to show this. However, could some sort of timing issue somehow show that it had the previous day's figures past midnight, then it saw them as going back to 0 during that time period as being negative use?

It isn't just the Load Energy Today entity that seems to be off either. The Battery Charge and Consumed Solar are also massively off:
[image]

gcoan, 2025-01-02:
There are at least a couple of problems here:

1. When you reboot HA the givtcp sensors can 'bounce' giving false readings that you have to manually resolve by using Developer tools/statistics and adjusting the sensor history at the time of the reboot (these are obvious, big negative or positive changes to the sensor value)

2. Timing differences between the reset of the sensors on the inverter and your HA energy dashboard can cause the midnight spikes.  GivTCP3 seems more susceptible to this

Instead of using the givtcp today sensors, use the givtcp total sensors, these work just as well in the Energy dashboard and resolve number 2, and to a lesser extent number 1 although you may still have to do manual adjustments of the sensor statistics after a HA reboot

cbrinks17, 2025-01-05:
I ended up going back to 2.4.9. As discussed in the [other thread](https://github.com/britkat1980/giv_tcp/issues/252), GivTCP does not record the battery totals on the main device, and I didn't see any entities that were providing accurate total charge/discharges on the others. Hopefully someone else with an AIO can chime in to see if they have the battery total sensors.