britkat1980/ha-addons #45: Missing Battery Pause Mode Entity

wulliemc73, 2025-02-23:
I have had my inverter replaced and had cause to reinstall GivTCP 2.4.9.  In the main its all fine, but I now have a missing entity:

[image]

Entity is givtcp_xxx_battery_pause_mode

Its still available in the remote control section of my GE portal, so not sure why it isn't coming through on GivTCP:

[image]

If you could please look into this, it would be much appreciated.

The inverter is a GIV-HY5.0, software version D0.309-A0.309.

gcoan, 2025-02-23:
Battery pause wasn't available in the original firmware.  Have you got 'use old firmware' set to on in GivTCP configuration (or might be  'new firmware', can't remember)

wulliemc73, 2025-02-24:
That's interesting that you mention that.  When I was comparing my v2.4.9 with someone else's, they looked different, and he had a toggle about firmware and if it as an AIO, and I didn't.  Here is s screenshot of mine:

[image]

Assuming that could very well be the problem, where would I be able to get a the version of 2.4.9 that asks question on the firmware?  Or is there anything I can insert in the YAML that achieves the same outcome?

gcoan, 2025-02-24:
> toggle about firmware and if it as an AIO, and I didn't. Here is s screenshot of mine:

I've personally never seen those configuration options, is this a dev version of 2.4.9?
My recollection is that there was only the old/new firmware switch. 

Just curious why you're using givtcp 2 when v3 has been out for a while and is where all the development effort is. Upgrading might resolve the problem?

If you stick with v2, might be possible to edit the json configuration file directly but it looks like you have different options. 

> The inverter is a GIV-HY5.0, software version D0.309-A0.309.

Just to check, this isn't a 3 phase inverter is it? It's a single phase 5kw hybrid?

gcoan, 2025-02-24:
> Not a dev version that I know of, I re-downloaded it from github over the weekend having completely removed it, from here:
> 
> https://github.com/GivEnergy/giv_tcp

That may be your problem.  The GivEnergy version of givtcp hasn't been maintained for a long time.

You need to install GivTCP v3 from here: 

https://github.com/britkat1980/ha-addons

or if its v2, from here:

https://github.com/britkat1980/giv_tcp

There have been problems for some people (including me) in upgrading from v2 to v3, the battery entities were not populating and got 'duplicate id' errors from MQTT in the Home Assistant logs.  This is fixable and if thats the problem you had I can help you out.

wulliemc73, 2025-02-24:
Apologies, I re-checked, this is the repository I have installed from:

https://github.com/britkat1980/giv_tcp

I checked the YAML in the Addon Configuration page and changed:

INVERTOR_AC_1: true to INVERTOR_AC_1: false and the Pause options now show up

I might take you up on your offer ref v3 for the long term though because my pkl files all only have 1 entry in them and its an error message, not populated as you might expect.

wulliemc73, 2025-02-24:
Think this can be closed now.  Raising this issue has made me realise, I had the question all along, just without the words.  Updating the YAML changed the toggle compared to my screenshot a few posts ago:

[image]

closed 2025-02-24 by wulliemc73

gcoan, 2025-02-24:
> my pkl files all only have 1 entry in them and its an error message, not populated as you might expect.

Glad you're sorted

The pkl files are cache files so you can safely delete them and restart the add on and they should be created again