britkat1980/ha-addons #46: Can't install GivTCP v3

dshipp, 2025-03-04:
Following the instructions here: https://github.com/britkat1980/giv_tcp?tab=readme-ov-file

Once I've added the repository https://github.com/britkat1980/ha-addons to the add-on store, I can see the three GivTCP-3 add-ons.

When I try to install GivTCP, I get the following error:

`Can't install givtcp.docker.scarf.sh/britkat/giv_tcp-ma:3.0.4: 404 Client Error for http+docker://localhost/v1.47/images/givtcp.docker.scarf.sh/britkat/giv_tcp-ma:3.0.4/json: Not Found ("No such image: givtcp.docker.scarf.sh/britkat/giv_tcp-ma:3.0.4")`

It appears that the docker container image for GivTCP 3.0.4 is not located where it is expecting it.
Similarly if I try GivTCP-Beta:

`Can't install givtcp.docker.scarf.sh/britkat/giv_tcp-beta:3.1.2: 404 Client Error for http+docker://localhost/v1.47/images/givtcp.docker.scarf.sh/britkat/giv_tcp-beta:3.1.2/json: Not Found ("No such image: givtcp.docker.scarf.sh/britkat/giv_tcp-beta:3.1.2")`

britkat1980, 2025-03-04:
What machine are you installing on. Particularly what architecture (rpi or Linux machine)

dshipp, 2025-03-05:
It's a Raspberry Pi 3 (32-bit)

dshipp, 2025-03-05:
After some further digging (thanks Claude.ai!) it looks like my problem is running the 32bit OS for my Raspberry Pi. It looks like only 64-bit x86 and ARM architectures are supported in the giv_tcp:3.0.4 image.

@britkat1980 how hard would it be to add 32bit support like the old giv_tcp had? Or should I look at upgrading my Home Assistant host? Or am I barking up the wrong tree? Thanks.

britkat1980, 2025-03-05:
At one point int he dev some features required 64bit arch. However I don’t think that is currently true. Let me try a new build this evening and you can give it a go!

dshipp, 2025-03-06:
Thanks. Did you have any luck with the 32bit build? No worries if not, I'm sure I can re-home my HA instance.

britkat1980, 2025-03-06:
I left a build running overnight but it failed. Normally only takes 15mins for a full build so I wasn’t hopeful!
I think that there is a 64bit dependency in v3 so I don’t think I can make it backward compatible.

dshipp, 2025-03-06:
No problem. I appreciate you helping me diagnose my issue and giving it a try. Nice to have an excuse to upgrade my HA stack!

closed 2025-03-06 by dshipp