britkat1980/ha-addons #48: GivTCP Beta 3.1.2 & AIO Gateway

campbellcopeland, 2025-03-20:
Error in logs when single AIO and Gateway configured - Gateway has error in log

025-03-20 09:55:24,456 - GivTCP2 - read        -  [ERROR   ] - Error processing Gateway data: ((<class 'UnboundLocalError'>, UnboundLocalError("cannot access local variable 'power_flow_output' where it is not associated with a value"), <traceback object at 0x7ff26e130900>), 1693)
2025-03-20 09:55:24,457 - GivTCP2 - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7ff26e132b80>), 1932)
2025-03-20 09:55:24,457 - GivTCP2 - read        -  [ERROR   ] - runAll2 Error processing registers: ('KeyError', 'read.py', 2113)

SirSixx, 2025-04-13:
Can confirm error is still there in 3.1.3

> 2025-04-13 10:30:05,623 -  - read        -  [ERROR   ] - Error processing Gateway data: ((<class 'UnboundLocalError'>, UnboundLocalError("cannot access local variable 'power_flow_output' where it is not associated with a value"), <traceback object at 0x7f7e661b00>), 1705)
> 2025-04-13 10:30:05,623 -  - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7f7e654780>), 1943)
> 2025-04-13 10:30:05,623 -  - read        -  [ERROR   ] - runAll2 Error processing registers: ('KeyError', 'read.py', 2124)

I believe this was previously an issue that was fixed and is related to a AIO where there is a gateway with no battery.

campbellcopeland, 2025-04-13:
Seems to be fixed in 3.1.4-beta and is related to a Gateway with single AIO battery from what I understand.

closed 2025-04-13 by campbellcopeland