britkat1980/ha-addons #51: Unable to set "Export (Paused)" from automation or Developer tools - Actions

godfreym29, 2025-05-01:
Trying to set `Export (Paused)` in a automation does not execute and a error is generated in the log.

`[140312419422496] Option Export (paused) is not valid for entity select.givtcp_ce2133gxxx_mode, valid options are: Eco, Eco (Paused), Timed Demand, Timed Export, Unknown, Export (Paused)`

I have tried it using Developer tools - Actions and although the tool shows a green tick, it errors in the log.

HA is up to date
GivTCP is v3.1.6
I have a Givenergy AC3 and 8.2 kWh battery, both are running the latest firmware

I have tried executing each of the GivTCP Control Mode options using the Developer tools and all execute OK except for `Export (Paused)`
This is the write_log_inv_1.log output
`2025-05-01 09:08:48,222 - write - [INFO] - Setting Eco mode was a success`
`2025-05-01 09:09:05,231 - write - [INFO] - Setting dynamic mode was a success`
`2025-05-01 09:09:22,991 - write - [INFO] - Setting Timed Demand mode was a success`
`2025-05-01 09:09:37,734 - write - [INFO] - Setting Timed Export mode was a success`
`2025-05-01 09:09:52,080 - write - [ERROR] - Invalid Mode requestedExport (Paused)`

Debug logs
[log_inv_1.log](https://github.com/user-attachments/files/19998078/log_inv_1.log)

[write_log_inv_1.log](https://github.com/user-attachments/files/19998088/write_log_inv_1.log)

britkat1980, 2025-05-01:
Ok thanks, this is because it was only recently implemented as a read-only status. I never implemented it as a control state. What behaviour do you want it to give you, and I’ll look at implementing.

godfreym29, 2025-05-01:
On that I'm not sure.
I'm on Octopus Intelligent Flux and they are using it fairly often and sometimes I would like to be able to reset it.
Is there anything I can look at to help with the answer?

gcoan, 2025-05-01:
> I'm on Octopus Intelligent Flux and they are using it fairly often and sometimes I would like to be able to reset it.

If you are on OIF then any changes you make to the inverter will be overwritten by Octopus pretty quickly.

I thought Export (Paused) wasn't an explicit control mode you could set, you set mode Export and a battery reserve eg 20% and when the SoC reaches that reserve the mode automatically changes to Export (Paused)

godfreym29, 2025-05-01:
I don't see a GivTCP Control Mode option for Export, do you perhaps mean Timed Export?

gcoan, 2025-05-01:
> I don't see a GivTCP Control Mode option for Export, do you perhaps mean Timed Export?

yes correct