britkat1980/ha-addons #52: Versie 3.1.7 does not work with 3-phase hybrid inverter

lancer73, 2025-05-02:
After updating to this version the log shows:
```
2025-05-02 13:49:18,621 - GivTCP - read        -  [INFO    ] - Starting watch_plant loop...
2025-05-02 13:49:18,621 - GivTCP - GivLUT      -  [CRITICAL] - Opening Modbus Connection to 192.168.107.108
2025-05-02 13:49:18,622 - GivTCP - read        -  [CRITICAL] - Detecting inverter characteristics...
2025-05-02 13:49:37,357 - GivTCP - read        -  [ERROR   ] - Error processing Three Phase data: ((<class 'givenergy_modbus_async.exceptions.ConversionError'>, ConversionError('generation: [0]: 0 is not a valid Generation'), <traceback object at 0x7f2677130380>), 1766)
2025-05-02 13:49:37,357 - GivTCP - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7f2676d9b480>), 1955)
2025-05-02 13:49:37,357 - GivTCP - read        -  [ERROR   ] - runAll2 Error processing registers: ('KeyError', 'read.py', 2136)
2025-05-02 13:50:10,574 - GivTCP - read        -  [ERROR   ] - Error processing Three Phase data: ((<class 'givenergy_modbus_async.exceptions.ConversionError'>, ConversionError('generation: [0]: 0 is not a valid Generation'), <traceback object at 0x7f26771307c0>), 1766)
2025-05-02 13:50:10,574 - GivTCP - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7f2677130800>), 1955)
2025-05-02 13:50:10,574 - GivTCP - read        -  [ERROR   ] - runAll2 Error processing registers: ('KeyError', 'read.py', 2136)
2025-05-02 13:50:49,297 - GivTCP - read        -  [ERROR   ] - Error processing Three Phase data: ((<class 'givenergy_modbus_async.exceptions.ConversionError'>, ConversionError('generation: [0]: 0 is not a valid Generation'), <traceback object at 0x7f2676d9b8c0>), 1766)
2025-05-02 13:50:49,298 - GivTCP - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7f2677130800>), 1955)
2025-05-02 13:50:49,298 - GivTCP - read        -  [ERROR   ] - runAll2 Error processing registers: ('KeyError', 'read.py', 2136)
2025-05-02 13:51:27,259 - GivTCP - read        -  [ERROR   ] - Error processing Three Phase data: ((<class 'givenergy_modbus_async.exceptions.ConversionError'>, ConversionError('generation: [0]: 0 is not a valid Generation'), <traceback object at 0x7f2677131240>), 1766)
2025-05-02 13:51:27,260 - GivTCP - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7f2677133f80>), 1955)
2025-05-02 13:51:27,260 - GivTCP - read        -  [ERROR   ] - runAll2 Error processing registers: ('KeyError', 'read.py', 2136)
2025-05-02 13:52:02,483 - GivTCP - read        -  [ERROR   ] - Error processing Three Phase data: ((<class 'givenergy_modbus_async.exceptions.ConversionError'>, ConversionError('generation: [0]: 0 is not a valid Generation'), <traceback object at 0x7f2676d9bac0>), 1766)
2025-05-02 13:52:02,483 - GivTCP - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7f2676d9be00>), 1955)
2025-05-02 13:52:02,483 - GivTCP - read        -  [ERROR   ] - runAll2 Error processing registers: ('KeyError', 'read.py', 2136)
2025-05-02 13:52:38,445 - GivTCP - read        -  [ERROR   ] - Error processing Three Phase data: ((<class 'givenergy_modbus_async.exceptions.ConversionError'>, ConversionError('generation: [0]: 0 is not a valid Generation'), <traceback object at 0x7f2677130cc0>), 1766)
2025-05-02 13:52:38,445 - GivTCP - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7f2677133b00>), 1955)
2025-05-02 13:52:38,445 - GivTCP - read        -  [ERROR   ] - runAll2 Error processing registers: ('KeyError', 'read.py', 2136)

```
And no MQTT entity is updated. The webinterface of the addon just hangs.

britkat1980, 2025-05-02:
Please try Beta3.1.8 to see if its fixed

lancer73, 2025-05-02:
Different error now

```
2025-05-02 20:10:41,493 - startup     -  [INFO    ] - Network Found: 192.168.107.14/24
2025-05-02 20:10:41,493 - startup     -  [INFO    ] - Network Found: 192.168.14.14/24
2025-05-02 20:10:41,493 - startup     -  [INFO    ] - Network Found: 192.168.106.14/24
2025-05-02 20:10:43,498 - startup     -  [INFO    ] - Searching for Inverters
2025-05-02 20:10:43,498 - startup     -  [INFO    ] - Scanning network for GivEnergy Devices...
2025-05-02 20:10:55,918 - client      -  [INFO    ] - Connection established to 192.168.107.108:8899
2025-05-02 20:10:55,919 - client      -  [INFO    ] - Detecting plant
2025-05-02 20:11:03,845 - client      -  [INFO    ] - Plant Detected
2025-05-02 20:11:12,878 - client      -  [INFO    ] - Connection established to 192.168.107.108:8899
2025-05-02 20:11:12,878 - client      -  [INFO    ] - Detecting plant
2025-05-02 20:11:14,045 - client      -  [INFO    ] - Plant Detected
2025-05-02 20:11:21,994 - client      -  [INFO    ] - Connection established to 192.168.107.108:8899
2025-05-02 20:11:21,994 - client      -  [INFO    ] - Detecting plant
2025-05-02 20:11:23,155 - client      -  [INFO    ] - Plant Detected
2025-05-02 20:11:28,678 - client      -  [INFO    ] - Connection established to 192.168.107.108:8899
2025-05-02 20:11:28,678 - client      -  [INFO    ] - Detecting plant
2025-05-02 20:11:29,836 - client      -  [INFO    ] - Plant Detected
2025-05-02 20:11:37,784 - client      -  [INFO    ] - Connection established to 192.168.107.108:8899
2025-05-02 20:11:37,784 - client      -  [INFO    ] - Detecting plant
2025-05-02 20:11:38,945 - client      -  [INFO    ] - Plant Detected
2025-05-02 20:11:44,968 - client      -  [INFO    ] - Connection established to 192.168.107.108:8899
2025-05-02 20:11:44,968 - client      -  [INFO    ] - Detecting plant
2025-05-02 20:11:49,637 - client      -  [INFO    ] - Plant Detected
2025-05-02 20:11:59,932 - startup     -  [INFO    ] - Setting up invertor: 1
2025-05-02 20:11:59,953 - startup     -  [INFO    ] - ==============================================================
2025-05-02 20:11:59,954 - startup     -  [INFO    ] - ====             Web Gui Config is at                     ====
2025-05-02 20:11:59,954 - startup     -  [INFO    ] - ====     http://192.168.106.14:8099/config.html             ====
2025-05-02 20:11:59,954 - startup     -  [INFO    ] - ==============================================================
2025-05-02 20:11:59,954 - startup     -  [INFO    ] - Running Invertor 1 (TC2337G165) read loop every 30/120s
[2025-05-02 20:12:00 +0200] [150] [INFO] Starting gunicorn 23.0.0
[2025-05-02 20:12:00 +0200] [150] [INFO] Listening at: http://0.0.0.0:6350 (150)
[2025-05-02 20:12:00 +0200] [150] [INFO] Using worker: sync
[2025-05-02 20:12:00 +0200] [161] [INFO] Booting worker with pid: 161
[2025-05-02 20:12:00 +0200] [160] [INFO] Starting gunicorn 23.0.0
[2025-05-02 20:12:00 +0200] [160] [INFO] Listening at: http://0.0.0.0:6345 (160)
[2025-05-02 20:12:00 +0200] [160] [INFO] Using worker: sync
[2025-05-02 20:12:00 +0200] [162] [INFO] Booting worker with pid: 162
[2025-05-02 20:12:00 +0200] [163] [INFO] Booting worker with pid: 163
[2025-05-02 20:12:00 +0200] [164] [INFO] Booting worker with pid: 164
2025-05-02 20:12:00,426 - GivTCP - read        -  [INFO    ] - Starting watch_plant loop...
2025-05-02 20:12:00,426 - GivTCP - GivLUT      -  [CRITICAL] - Opening Modbus Connection to 192.168.107.108
2025-05-02 20:12:00,427 - GivTCP - read        -  [CRITICAL] - Detecting inverter characteristics...
2025-05-02 20:12:22,417 - GivTCP - read        -  [ERROR   ] - Error processing Three Phase data: ((<class 'TypeError'>, TypeError("Converter.generation() missing 1 required positional argument: 'fw'"), <traceback object at 0x7faf29ac7500>), 1766)
2025-05-02 20:12:22,417 - GivTCP - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7faf29ac7e00>), 1955)
2025-05-02 20:12:22,417 - GivTCP - read        -  [ERROR   ] - runAll2 Error processing registers: ('KeyError', 'read.py', 2136)
2025-05-02 20:13:00,382 - GivTCP - read        -  [ERROR   ] - Error processing Three Phase data: ((<class 'TypeError'>, TypeError("Converter.generation() missing 1 required positional argument: 'fw'"), <traceback object at 0x7faf29eebc80>), 1766)
2025-05-02 20:13:00,383 - GivTCP - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7faf29eebbc0>), 1955)
2025-05-02 20:13:00,384 - GivTCP - read        -  [ERROR   ] - runAll2 Error processing registers: ('KeyError', 'read.py', 2136)
2025-05-02 20:13:35,343 - GivTCP - read        -  [ERROR   ] - Error processing Three Phase data: ((<class 'TypeError'>, TypeError("Converter.generation() missing 1 required positional argument: 'fw'"), <traceback object at 0x7faf29eebec0>), 1766)
2025-05-02 20:13:35,343 - GivTCP - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7faf29ee8040>), 1955)
2025-05-02 20:13:35,344 - GivTCP - read        -  [ERROR   ] - runAll2 Error processing registers: ('KeyError', 'read.py', 2136)
2025-05-02 20:14:10,555 - GivTCP - read        -  [ERROR   ] - Error processing Three Phase data: ((<class 'TypeError'>, TypeError("Converter.generation() missing 1 required positional argument: 'fw'"), <traceback object at 0x7faf29ee85c0>), 1766)
2025-05-02 20:14:10,556 - GivTCP - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7faf29eeb5c0>), 1955)
2025-05-02 20:14:10,557 - GivTCP - read        -  [ERROR   ] - runAll2 Error processing registers: ('KeyError', 'read.py', 2136)
```

britkat1980, 2025-05-02:
3.1.9…

closed 2025-12-29 by lancer73